PiSCSI/piscsi #1: CD-ROM Emulation doesn't work with stock Apple drivers

akuker, 2020-07-08:
The "stock" version of Apple's CD-ROM driver for MacOS 8.x (and probably all versions of "classic" Mac OSs) does not recognize the emulated CD-ROM of RaSCSI. At startup, the CD-ROM driver will send an unsupported command that is not supported by RaSCSI. 

I believe this check was done to make sure that only CD-ROMs with Apple firmware were used.

There are a couple work-arounds:
- Install ResExcellence's CD-ROM driver hack (https://web.archive.org/web/20040404121044/http://www.resexcellence.com/hack_html_99/12-21-98.shtml)
- [Untested] Use a 3rd party CD-ROM driver. The CD driver that was distributed with Toast may be a good one to try.

erichelgeson, 2020-10-12:
FYI https://macintoshgarden.org/apps/cd-sunrise-v22b did not work with rascsi cd emulation

rdmark, 2021-08-17:
For the record, the Toast CD Reader driver from https://macintoshgarden.org/apps/astarte-toast-256 does not work either.  The extension just fails to load when testing on System 7.1 (Pro and P6) as well as Mac OS 7.6.1, even though a RaSCSI CD drive is attached.

rdmark, 2021-08-17:
One observation right now, is that since ResExcellence's hack is the only known (?) solution, and the oldest version of Apple CD-ROM that the hack is proven to work with is 5.4.2, effectively the oldest Mac OS version that can be used right now with RaSCSI CD-ROM is 8.1.
Today I spent a few hours trying to hack earlier Apple CD-ROM drivers without success. I also tried to use 5.4.2 with older system software versions, also without success.

rdmark, 2021-08-18:
I've got some good news:  After some research and testing today, I am glad to announce that I've found two 3rd party drivers that work well with RaSCSI:  RapidCD and CD/DVD Drive TuneUp 2.0.3
See my detailed test results here:  http://shorturl.at/wKNY9

My favorite is RapidCD, since it mounts CDs super fast, also can handle the DOS partitions, and is the only one that works on my stock Mac SE.  It even works on System 6.0.7, which blew me away!

uweseimet, 2021-08-21:
@rdmark It would be helpful to have a log on trace level for those cases where software fails to use the emulated devices.

rdmark, 2021-08-22:
@uweseimet This is an example when attaching an image and the system not being able to read the disk.  Hacked Apple CD ROM 5.3.1, System 7.1.1 on a Mac SE.

[2021-08-22 07:07:51.674] [info] Validating: cmd=ATTACH, id=4, unit=0, type=SCCD, filename='/home/pi/images/71.iso', device name='', params=''
[2021-08-22 07:07:51.676] [info] Executing: cmd=ATTACH, id=4, unit=0, type=SCCD, filename='/home/pi/images/71.iso', device name='', params=''
[2021-08-22 07:07:51.679] [info] Media capacity for image file '/home/pi/images/71.iso': 39067 blocks
[2021-08-22 07:07:51.681] [info] Added new SCCD device, ID: 4 unit: 0
[2021-08-22 07:07:51.810] [info] +----+----+------+-------------------------------------
[2021-08-22 07:07:51.813] [info] | ID | UN | TYPE | DEVICE STATUS
[2021-08-22 07:07:51.816] [info] +----+----+------+-------------------------------------
[2021-08-22 07:07:51.816] [info] |  3 |  0 | SCHD | /home/pi/images/600MB.hda
[2021-08-22 07:07:51.824] [info] |  4 |  0 | SCCD | /home/pi/images/71.iso (WRITEPROTECT)
[2021-08-22 07:07:51.825] [info] +----+----+------+-------------------------------------
[2021-08-22 07:07:52.668] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=4 (with device)
[2021-08-22 07:07:52.668] [trace] virtual void SASIDEV::Command() Command Phase
[2021-08-22 07:07:52.668] [trace] virtual void SCSIDEV::Execute() Execution phase command $00
[2021-08-22 07:07:52.669] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() ID 4 received CmdTestUnitReady ($00)
[2021-08-22 07:07:52.669] [trace] void SASIDEV::CmdTestUnitReady() TEST UNIT READY Command
[2021-08-22 07:07:52.669] [trace] Setting status: Sense Key: $06, ASC: $28, ASCQ: $00
[2021-08-22 07:07:52.669] [trace] BOOL Disk::CheckReady() Disk in needs attention
[2021-08-22 07:07:52.669] [trace] virtual void SCSIDEV::Error(ERROR_CODES::sense_key, ERROR_CODES::asc) Sense Key and ASC for subsequent REQUEST SENSE: $00, $00
[2021-08-22 07:07:52.669] [trace] virtual void SCSIDEV::Error(ERROR_CODES::sense_key, ERROR_CODES::asc) Error (to status phase)
[2021-08-22 07:07:52.669] [trace] void SASIDEV::Status() Status Phase $02
[2021-08-22 07:07:52.669] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-08-22 07:07:52.670] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
[2021-08-22 07:07:52.670] [trace] void SASIDEV::MsgIn() Starting Message in phase
[2021-08-22 07:07:52.670] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-08-22 07:07:52.670] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-08-22 07:07:52.670] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-08-22 07:07:52.670] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[2021-08-22 07:07:52.670] [trace] virtual void SCSIDEV::BusFree() Bus free phase
[2021-08-22 07:07:52.672] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=4 (with device)
[2021-08-22 07:07:52.672] [trace] virtual void SASIDEV::Command() Command Phase
[2021-08-22 07:07:52.673] [trace] virtual void SCSIDEV::Execute() Execution phase command $03
[2021-08-22 07:07:52.673] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() ID 4 received CmdRequestSense ($03)
[2021-08-22 07:07:52.673] [trace] void SASIDEV::CmdRequestSense() REQUEST SENSE Command
[2021-08-22 07:07:52.673] [trace] virtual int Disk::RequestSense(const DWORD*, BYTE*) size of data = 18
[2021-08-22 07:07:52.673] [trace] Setting status: Sense Key: $00, ASC: $00, ASCQ: $00
[2021-08-22 07:07:52.673] [trace] void SASIDEV::CmdRequestSense() Sense Key $06, ASC $28
[2021-08-22 07:07:52.673] [trace] void SASIDEV::DataIn() Going into Data-in Phase
[2021-08-22 07:07:52.673] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 18
[2021-08-22 07:07:52.674] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
[2021-08-22 07:07:52.674] [trace] void SASIDEV::Status() Status Phase $00
[2021-08-22 07:07:52.674] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-08-22 07:07:52.674] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
[2021-08-22 07:07:52.674] [trace] void SASIDEV::MsgIn() Starting Message in phase
[2021-08-22 07:07:52.674] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-08-22 07:07:52.674] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-08-22 07:07:52.675] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-08-22 07:07:52.675] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[2021-08-22 07:07:52.675] [trace] virtual void SCSIDEV::BusFree() Bus free phase
[2021-08-22 07:07:52.676] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=4 (with device)
[2021-08-22 07:07:52.676] [trace] virtual void SASIDEV::Command() Command Phase
[2021-08-22 07:07:52.677] [trace] virtual void SCSIDEV::Execute() Execution phase command $00
[2021-08-22 07:07:52.677] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() ID 4 received CmdTestUnitReady ($00)
[2021-08-22 07:07:52.677] [trace] void SASIDEV::CmdTestUnitReady() TEST UNIT READY Command
[2021-08-22 07:07:52.677] [trace] Setting status: Sense Key: $00, ASC: $00, ASCQ: $00
[2021-08-22 07:07:52.677] [trace] BOOL Disk::CheckReady() Disk is ready!
[2021-08-22 07:07:52.677] [trace] void SASIDEV::Status() Status Phase $00
[2021-08-22 07:07:52.678] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[... cut ...]
[2021-08-22 07:07:52.679] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[2021-08-22 07:07:52.679] [trace] virtual void SCSIDEV::BusFree() Bus free phase
[2021-08-22 07:07:52.680] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=4 (with device)
[2021-08-22 07:07:52.680] [trace] virtual void SASIDEV::Command() Command Phase
[2021-08-22 07:07:52.681] [trace] virtual void SCSIDEV::Execute() Execution phase command $43
[2021-08-22 07:07:52.681] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() ID 4 received CmdReadToc ($43)

rdmark, 2021-08-22:
Thanks to Stinkerton18 on Discord, we now have a patched version of the Apple CD-ROM driver that works at least as far back as on System 6.0.7!  I've uploaded both to the Garden: https://macintoshgarden.org/apps/apple-cd-rom-extension-patched

BTW, I've fallen out of love slightly with RapidCD.  While highly compatible, it's a major memory hog.  On system 7.1.1 one attached CD drive takes up 680kb, while Apple's driver only uses 110kb per drive.  On a memory limited system like a 4MB SE, RapidCD is almost unusable in real life.

uweseimet, 2021-08-22:
@rdmark Regarding the log above, does it stop (no further logging) after the CmdReadToc command? I would have expected more output.

rdmark, 2021-08-22:
@uweseimet Oh no, a loop of similar looking sequences of log messages keep going at a very rapid pace while using a CD device.  I can give you a few thousand lines worth if it helps!

uweseimet, 2021-08-22:
@rdmark  :-)  Please just some more lines until it is obvious that there is a loop. If I'm not mistaken CmdReadToc is only called once in the log above, i.e. the loop does not seem to have started.

rdmark, 2021-08-22:
[cd_trace.txt](https://github.com/akuker/RASCSI/files/7028206/cd_trace.txt)
Here you go!

uweseimet, 2021-08-22:
Now one can see that a MODE SELECT fails (see logfile excerpt below). Looks as if the CD-ROM software is trying to configure the device but RaSCSI doesn't allow that.
Maybe it is sufficient to just return an OK status in this case, without actually doing something for this specific MODE SELECT. It may be worth trying this at some time in the future.

```
[2021-08-22 22:40:48.717] [trace] virtual void SCSIDEV::Execute() Execution phase command $15
[2021-08-22 22:40:48.717] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() ID 5 received CmdModeSelect ($15)
[2021-08-22 22:40:48.717] [trace] void SCSIDEV::CmdModeSelect() MODE SELECT Command
[2021-08-22 22:40:48.717] [trace] void SASIDEV::DataOut() Data out phase
[2021-08-22 22:40:48.717] [trace] virtual void SCSIDEV::Receive()
[2021-08-22 22:40:48.717] [trace] virtual void SCSIDEV::Receive() length was != 0
[2021-08-22 22:40:48.717] [trace] virtual void SCSIDEV::Receive()
[2021-08-22 22:40:48.717] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 7 (dataout)
[2021-08-22 22:40:48.717] [trace] Setting status: Sense Key: $05, ASC: $26, ASCQ: $00
[2021-08-22 22:40:48.718] [trace] virtual void SCSIDEV::Error(ERROR_CODES::sense_key, ERROR_CODES::asc) Sense Key and ASC for subsequent REQUEST SENSE: $00, $00
[2021-08-22 22:40:48.718] [trace] virtual void SCSIDEV::Error(ERROR_CODES::sense_key, ERROR_CODES::asc) Error (to status phase)
[2021-08-22 22:40:48.718] [trace] void SASIDEV::Status() Status Phase $02
[2021-08-22 22:40:48.718] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-08-22 22:40:48.718] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
[2021-08-22 22:40:48.718] [trace] void SASIDEV::MsgIn() Starting Message in phase
[2021-08-22 22:40:48.718] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-08-22 22:40:48.718] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-08-22 22:40:48.718] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-08-22 22:40:48.718] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[2021-08-22 22:40:48.718] [trace] virtual void SCSIDEV::BusFree() Bus free phase
[2021-08-22 22:40:48.719] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=5 (with device)
[2021-08-22 22:40:48.720] [trace] virtual void SASIDEV::Command() Command Phase
[2021-08-22 22:40:48.720] [trace] virtual void SCSIDEV::Execute() Execution phase command $03
[2021-08-22 22:40:48.720] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() ID 5 received CmdRequestSense ($03)
```

uweseimet, 2021-08-23:
@rdmark The current uweseimet_develop branch contains additional debug output, which should reveal which code page MODE SELECT wants to change. Would be great if you provided updated logs whenever you have the time for it.

rdmark, 2021-08-23:
@uweseimet Do I need to compile rascsi with debug symbols to get the right level of debug output?

rdmark, 2021-08-23:
@uweseimet trace using b40c6b1 on my Mac SE, System 6.0.7, Apple CD-ROM driver.  Up through booting the system off a non-RaSCSI hard drive, the system complaining that the CD file system can't be read, and me hitting the Eject button.
[cd_trace2.txt](https://github.com/akuker/RASCSI/files/7034796/cd_trace2.txt)

uweseimet, 2021-08-23:
@rdmark Thank you for providing updated logs. At least from the new logs we know now which mode page the driver tries to use:

```
[2021-08-23 21:48:02.151] [debug] virtual bool Disk::Dispatch(SCSIDEV*) received CmdModeSelect ($15)
[2021-08-23 21:48:02.151] [trace] MODE SELECT for unsupported page 21
```

When (and if) somebody starts investigating this issue thsi piece of information might be useful.

akuker, 2021-12-30:
There's also an issue with ModeSense6. When the Mac sends the ModeSense6 command, it specifies a size of 0x14 (20 decimal). This graphic shows that RaSCSI doesn't respond to the ModeSense6 for Page 1 request. It transitions to an error status 2.
[image]

The current RaSCSI code calculates a response of size 24. This is too big, and I don't think RaSCSI is calculating the response size appropriately. The response size should be 20 bytes (as requested by the Mac).... 4 bytes (for the mode sense data header) + 8 bytes (for the block descriptor) + 8 bytes (for the error recovery parameters).

[image]

[image]

[image]

@uweseimet - I'm curious about your opinions on the risk associated with this change.

akuker, 2021-12-30:
Another interesting observation - With the Apple CD drive I was using, I didn't ever see a ModeSelect6 (15h) command in the SCSI data capture. I'm going to do some more data captures from some other drives to make sure this wasn't a fluke. 

My wild theory? If the drivers get the data they expect from the Inquiry and/or ModeSense6 commands, it doesn't need to change anything using the ModeSelect command.

uweseimet, 2021-12-30:
@akuker Well, there is no risk if your change is compliant with the SCSI specification. If it is not the change will break a lot. In cases like this it is very important to check against the specification.

einstein95, 2022-06-16:
It appears you've got the wrong response for page 0x30 comparing it to the one in MAME (which does work with stock AppleCD drivers). See https://github.com/mamedev/mame/blob/badecba0e3aa372b75c738a4c53b2a7f8332064e/src/devices/bus/nscsi/cd.cpp#L427

rdmark, 2022-07-08:
@einstein95 Thanks for the tip! In addition to the MODE SENSE piece here, there seems to be an aspect of masquerading with a particular product string in the MAME solution: https://github.com/mamedev/mame/commit/c58e4b2a3e3624299da8362740c6c24a4c428abf

closed 2022-09-10 by uweseimet