PiSCSI/piscsi #2: RaSCSI crashes when MacOS reboots

akuker, 2020-07-08:
This issue was observed using MacOS 8.1, but likely applies to other versions as well.

When the Mac starts the reboot process, it will send a Flush command to RaSCSI that is not supported. This causes RaSCSI to immediately crash and will cause the Mac to freeze (before it reboots). 

To recover from this situation, you need to re-start RaSCSI.

    Received an invalid flush command 15!!!!!
    rascsi: disk.cpp:7864: void SASIDEV::FlushUnit(): Assertion `0' failed.
    make: *** [Makefile:108: run] Aborted

akuker, 2020-07-09:
Some observations:
While the drive is in DataOut mode, the initiator is sending a mode select command with the page code 0x08.

From void FASTCALL SASIDEV::FlushUnit().......
  Operation Code: [15]
  Logical Unit 0, PF 1, SP 0 [10]
  Reserved: 00
  Reserved: 00
  Parameter List Len 18
  Reserved: 00
  Ctrl Len: 00000000

From SCSIHD::ModeSelect()......
Received page code 8 with length 10
     08 0A 00 00 00 00 00 00 00 00 00 00

Mode Page Code 8 is defined in this old Sony interface manual, but the size of the command doesn't match.
http://www.bitsavers.org/pdf/sony/cdrom/Sony_CDU-541_SCSI_Interface_Manual_Mar1990.pdf

For now, updating RaSCSI to accept and ignore Mode Page Code 8 without throwing an error.

akuker, 2020-07-10:
I'm going to consider this fix sufficient for now. Issue #1 will need to clean-up the Mac driver-specific commands. Everything seems to still work when this mode select command is ignored (with the patched driver)

uweseimet, 2022-02-23:
@akuker Can you please verify whether this issue still exists? This issue looks very strange, and the work around is not such a good solution. (I have never seen the warning from the work-around, by the way.)
In order to test this you would have to remove the work-around in scsihd.cpp.

akuker, 2022-02-24:
> Can you please verify whether this issue still exists? This issue looks very strange, and the work around is not such a good solution.

This is an old one! I'm a little swamped at work right now, but will try it out as soon as I get a chance.

uweseimet, 2022-08-26:
@akuker @rdmark Half a year has gone ;-). I suggest to close this ticket and remove the related debug code. There were so many code changes anyway that it's not clear whether this ticket is still valid.

akuker, 2022-08-26:
No objections closing this. Thank you for doing some backlog grooming, @uweseimet !

closed 2022-08-26 by akuker