PiSCSI/piscsi #3: rasctl should be able shut down RaSCSI

akuker, 2020-07-08:
When running as a background service, there isn't an easy way too command RaSCSI to terminate. Suggest the following updates be made:
- A "-stop" command is added to rasctl that will command RaSCSI to terminate
- RaSCSI needs to be updated to exit gracefully when that command is received.

akuker, 2020-08-14:
Note: I started working this in with issue #16.

phrax0, 2021-05-07:
I don't know if this is in the dev tree, but if not, propose we start/stop the service with systemctl <start/stop> rascsi.  What are your thoughts on this?

akuker, 2021-05-07:
Sort of.... systemctl stop rascsi doesn't do a graceful exit. It just kills the process. So, rascsi could be in the middle of a transaction, or saving cached data back to disk. 

The idea is that systemctl would call a 'rasctl' command to shutdown rascsi, isntead of just killing the process.

closed 2021-10-17 by rdmark