PiSCSI/piscsi #4: RaSCSI file handling improvments

akuker, 2020-07-08:
Hard Disk Images: "Apple" hard disk images must be named ".hda" (lower case). This should be opened up a little more to accept .HDA and potentially .img files.
CD-ROM images: CD rom images MUST be named ".iso". This should be expanded to allow for .ISO and also .toast or .TOAST.

phrax0, 2021-05-30:
It looks like rasctl.cpp was updated to use xstrcasecmp(ext, "hda") for the different extension, so I think the case sensitivity is sorted.

rdmark, 2021-08-20:
As of b32f324 it's also possible to use CD ROM images with at least .toast and .img file extensions that I have tested.

rdmark, 2021-09-15:
Tested with 8a3642b -- it's possible to attach images with file endings such as HDA, TOAST, and ISO, in addition to the lower case variants.

closed 2021-09-15 by rdmark