PiSCSI/piscsi #5: Create a web control interface for RaSCSI

akuker, 2020-07-08:
Create a web interface to control/interact with RaSCSI.
_**This web interface should work with old browsers! At a minimum, Netscape 4, but even older would be better.**_

A potential starting point:
https://github.com/ztto/rascsi-php

Some potential features:
* Add/remove drives
* Change removable media for CD-ROMs
* Be able to upload/manage CD-ROMs stored on the Pi
* Restart the RaSCSI service

Additional features:
* See usages statistics for the Pi, such as:
** Disk/memory/storage usage
** How many bytes transfered for each SCSI ID?
** Error logs?
* Be able to upload/manage CD-ROMs stored on the Pi

phrax0, 2020-08-15:
Just commenting that as of 2020-08-14, the development of the web interface currently includes many of the features noted in this issue.

akuker, 2020-08-16:
Agreed. I think this can be closed.

The following PHP interface is in place to allow web control:
https://github.com/akuker/RASCSI/tree/master/src/php

Setup instructions are here:
https://github.com/akuker/RASCSI/wiki/Web-Interface

closed 2020-08-16 by akuker