PiSCSI/piscsi #6: Add Zip drive capability (or other removable, write-able media support)

akuker, 2020-07-08:
It would be handy to be able to use write-able removable media, such as a Zip drive. Consider adding this as a future feature.

phrax0, 2020-08-15:
What would this look like, as compared to creating a 100MB or 250MB image and presenting that to the host computer?  I might be mis-understanding what this feature would provide. Maybe it's to make the host "see" the SCSI device as a Zip drive, as compared to a hard drive, optical drive, ditto, etc...

uweseimet, 2021-08-09:
@akuker Correct me if I am wrong: We just need a new device type (just like SCSI_HD) which has the 'removable' flag set, correct? Actually, except for the sector size the MO device type is such a device.

akuker, 2021-08-10:
You are correct - also probably need to have the correct vendor ID

uweseimet, 2021-08-10:
There is no need for a particular vendor ID because there are numerous removable SCSI devices types (Syquest, various MOs, ZIP, JAZ, ...), and the INQUIRY data indicate whether a drive is removable.
You would need a vendor ID in order to support proprietary ZIP/JAZ features like password-protected media, but I doubt that supporting this feature makes sense for RaSCSI. One would have ti implement code that encrypts the device file on the fly in order to protect its content.

closed 2021-09-15 by uweseimet