PiSCSI/piscsi #10: HW: version 1.5 of RaSCSI has ACT signal miswired

akuker, 2020-07-10:
Version 1.5 of the RaSCSI "68kmla" version has an incorrect connection. Pin 1 should be connected to the "ACT" signal, instead of the PI-ACK signal.

[image]

Workaround:

Cut the trace of ACT running to pin 1 of U3. Scrape off some solder mask on the side near the via.
[image]

Solder a jumper wire from the trace to Pin 7 of the Raspberry Pi GPIO header.
[image]

akuker, 2020-07-10:
Tagged as "won't fix". When the next batch of boards (if ever) are ordered, this should be fixed.

phrax0, 2020-08-15:
With the development of the 1.6 board addressing this issue, we can close #10 now.  Also, thanks for doing the 1.6 board akuker!

akuker, 2020-08-16:
Agreed!

closed 2020-08-16 by akuker