PiSCSI/piscsi #12: Add configuration mechanism

sfiera, 2020-08-08:
I was going to make a pull request to commit the systemd service from [Setup Instructions](https://github.com/akuker/RASCSI/wiki/Setup-Instructions#automatically-start-the-rascsi-software). However, that script passes `-HD1 /home/pi/HARDDISK.hda` which is likely to be incorrect. It would be useful to have a way to run RaSCSI with different configurations, without needing to change the systemd service.

My proposal: add a flag `--init script.sh`. RaSCSI would run the script after creating the monitor socket, and the script would execute rasctl commands to apply the desired configuration. That way, there wouldn’t be a need for a “config file format”.

akuker, 2020-08-08:
Great suggestion! Thanks!

sfiera, 2020-08-10:
In writing the getopt change, I found the existing code path for using a config file in baremetal mode. So option B would be to enable that code path in regular mode too. Thoughts?

akuker, 2020-08-10:
I was actually contemplating getting rid off the initialization configuration completely. This might sound crazy....

It bugs me that there is different syntax for the RaSCSI app vs the rasctl app. I’m thinking we could get rid of rascsi taking drive information at startup at all. Instead, you could pass it a configuration file (where we could add a lot more interesting configurations.... logging level, default “mode”... such as always using the Apple HD mode).

The alternate way of configuring it would be to have a shell script with a bunch of rasctl commands in it. We could even have a way to “save” the current config to a shell script. 

The whole command line parsing logic kind of bugs me.... I started down a deep, dark rabbit hole tonight... I’m hoping to have some restructuring of the code done soon to (hopefully) make the code easier to understand and maintain. 

Feel free to take a look at issue #16. It’s probably going to be a massive pull request.

sfiera, 2020-08-10:
Sounds reasonable to me. When I wrote the getopt change, my goal was to make it easy to add a --config or --init flag. Once that exists, I don't think users should need the existing flags.

akuker, 2020-08-14:
This was added in the GIMONS RaSCSI version 1.50. If the argc is == 2, he assumes its a file and will read in that config file. Not sure this is the best way to do it....

sfiera, 2020-08-14:
It could be done “compatibly” with #15 by assuming a config if no device ID has been set with `-d`. I think it would be better to explicitly use `-f config`, though.

uweseimet, 2022-08-26:
@akuker @rdmark One more ticket where I wonder whether it is still relevant after 2 years of inactivity.

akuker, 2022-08-26:
We basically have equivalent functionality through the web interface. The web interface is capable of saving off the configuration and re-loading it. So, if we look at the overall RaSCSI package, I think the goal of this has been accomplished.

closed 2022-08-26 by akuker