PiSCSI/piscsi #20: Merge in GIMONS 1.50 changes

akuker, 2020-08-13:
GIMONS recently released version 1.50 of the Japanese version. The 68kmla version of RaSCSI was based upon version 1.47.

1.50 appears to have some bug fixes and added "Burst" support.

Version 1.50: http://retropc.net/gimons/rascsi/rascsi.zip
Version 1.47: http://retropc.net/gimons/rascsi/rascsi147.zip

quorten, 2020-09-06:
Something that seems to have an implied answer here, but good to ask specifically.  Is there going to be a plan to have a streamlined common core code base or architecture that well supports both projects?  Maybe even possibility for new collaboration between the two?  It's good to see GIMONS sites are back on the modern Internet for one thing.

akuker, 2020-09-06:
Currently, there are no plans. I'm definitely open to the idea. But, I have a limitation that I don't speak or read Japanese. The original code was all in Japanese. 

I'm hoping that the two projects can share feature/bug fixes/etc, but for now, I think its going to be a manual process.

phrax0, 2021-05-31:
version 1.52 is out - http://retropc.net/gimons/rascsi/rascsi152.zip

rdmark, 2021-08-21:
@akuker Would it help this effort if you had a translated diff between 1.47 and 1.52?  I'd be happy to volunteer and make use of my mostly useless degree in Japanese. ;)
I glanced through a diff just now, and it seems to be less than a weekend's worth of translation between documentation and code comments.

phrax0, 2021-08-21:
This would be super awesome. Just to add, GIMONS is very responsive via email if you have questions. They are happy for the projects to co-exist, and I wouldn't hesitate to reach out if additional information was needed.

rdmark, 2021-08-21:
@phrax0 Any preferred diff format, before I dig into it?  Linux unified diff?

phrax0, 2021-08-22:
@rdmark I don't know enough to answer that question.  @akuker is the better person to ask.

rdmark, 2021-08-22:
Alright so digging a little deeper, this is my plan:

bin/ -> out of scope
doc/ -> translate a diff; only a few paragraphs in total
src/raspberrypi/*.h|*.cpp -> translate all files (a diff was just too noisy / too low in context to be translatable)
src/raspberrypi/BAREMETAL/*.h|*.c -> translate main.c which is the only file that seems to have Japanese code comments
src/x86k/* -> translate all files (very little content)

rdmark, 2021-08-23:
Some progress today:
I realized disk.cpp was just too ridiculously long to retranslate in full, so I went with a diff with 5 lines of context instead.
Here is a complete first pass of disk.cpp: https://github.com/rdmark/RASCSI-translate/blob/master/src/raspberrypi/disk.cpp.diff
It contains the implementation of "burst mode" which might interest y'all.

Also picked off some low-hanging fruit:
Fully translated BAREMETAL: https://github.com/rdmark/RASCSI-translate/blob/master/src/raspberrypi/BAREMETAL/main.c
Fully translated x68k/RASCTL: https://github.com/rdmark/RASCSI-translate/tree/master/src/x68k/RASCTL

Please let me know if this is helpful!

rdmark, 2021-08-23:
Translated the header file for disk too:  https://github.com/rdmark/RASCSI-translate/blob/master/src/raspberrypi/disk.h.diff

@uweseimet Since you're actively working in this part of the code, do you have an opinion on the best way to tackle evaluating / merging these upstream changes?

uweseimet, 2021-08-23:
@rdmark I don't think there are any untranslated texts anymore in disk.h. Merging something from old sources into the current develop tree is almost impossible, because there were really a lot of changes.

rdmark, 2021-08-25:
@uweseimet Well noted, this is also the impression I got from going through Gimon's code while glancing at the state of develop. In parts hardly recognizable as a fork of the other.

Regardless, I completed the translation of all code changes, just in case there are some valuable refactors, fixes or new functionality that can benefit this project. (e.g. Burst mode)

When comparing 1.52 with 1.47, I noted these major code organizational changes:
- cfilesystem.cpp|h renamed to fsdriver.cpp|h
- ctapdriver.cpp|h renamed to netdriver.cpp|h
- xm6.h renamed to rascsi.h
- Code in log.h, scsi.cpp|h have been folded into disk.cpp|h

rdmark, 2021-08-25:
As for doc/, I plan to retranslate those from scratch in https://github.com/akuker/RASCSI/issues/208

akuker, 2021-09-03:
> As for doc/, I plan to retranslate those from scratch in #208

Something to consider…. This project has evolved and the original Japanese documentation might not match the current implementation. 

@uwesiemet has been keeping the Man pages up to date. Is there anything in the originally documentation that isn’t talked about in the man pages (or on the wiki somewhere)

rdmark, 2021-09-03:
@akuker Eyeing through the Japanese text of the three txt files, I'd say there is a good amount of X68000 specific documentation that I don't see in the wiki.  Apart from that, this project has either diverged enough for it to be inapplicable, or it's basic stuff that's in man pages or the wiki.

rdmark, 2021-09-22:
One more step to resolve this ticket:  a PR to merge the C code and Makefile that builds a Human68k 'RASCTL.X' executable for X68000. Untested since I don't have a X68000 development environment set up. I still think it's worth to add this to the repo for completeness sake. 

https://github.com/akuker/RASCSI/pull/255

rdmark, 2021-09-25:
@akuker With the work done here at this point, do you think there anything else to be done in this ticket?

rdmark, 2021-10-01:
Adding rasdump/sasidump documentation in https://github.com/akuker/RASCSI/pull/295

rdmark, 2021-12-22:
Closing this ticket, with remaining unmerged feature "burst mode" as a Won't Fix: It is impossible to merge code as-is, and there unknown impact on various timing-sensitive platforms out there by introducing batched GPIO operations.

closed 2021-12-22 by rdmark