PiSCSI/piscsi #21: Unfork this repo so code searches in github work properly

eparadis, 2020-08-16:
Because this repo is a fork, code searches in github don't work.

[For example this one.](https://github.com/akuker/RASCSI/search?q=ALLOWED_FILE_TYPES&type=Code)

> Sorry, forked repositories are not currently searchable.

This process goes [something like this](https://nickymeuleman.netlify.app/blog/unfork-github-repo). Or [like this](https://stackoverflow.com/questions/38831301/how-to-un-fork-the-github-repository).

1. Rename your repository on GitHub
2. Import the repository

Pros:
- makes things easier to work with

Cons:
- seems to make some of the github meta info go away...

akuker, 2020-08-16:
I agree! Further down on the Stackoverflow link, it says that Github support sever the link without re-importing the repository. I submitted a ticket to them to see if they can do it that way.

akuker, 2020-08-16:
Well that was fast...... The Github support team made this change in less than 6 minutes!!

closed 2020-08-16 by akuker