PiSCSI/piscsi #23: Web interface should allow downloading of images

eparadis, 2020-08-18:
It seems like it would be useful to be able to download the images stored in the `image/` directory. It would complement the "create, upload, delete" functionality with "read".

There may be considerations of unmounting any images that are currently in use - I am not sure.

akuker, 2020-08-20:
This seems like a reasonable request. I guess right now, there is nothing stopping the user from doing bad things, like deleting images that are currently mounted. I'm not sure what rascsi will do, but it probably won't be good. So, I propose creating a sepearate ticket for adding some sort of protection mechanism for downloading/deleting/modifying images that are currently mounted.

rdmark, 2021-08-20:
I believe this has been implemented since.  Clicking on the button in the Size column allows you to download the image.  There is a little downwards arrow to indicate this.  Tested on b32f324

closed 2021-09-15 by rdmark