PiSCSI/piscsi #26: Create fast boot buildroot pre-built image

akuker, 2020-08-24:
The following github repo appears to have a minimal buildroot configuration that will boot in 1.5 seconds on a Raspberry Pi 3. We should create a Pi configuration that mimics this setup, but that is pre-configured to start rascsi.

https://github.com/romainreignier/minimal_raspberrypi_buildroot

Upon first boot, it a shell script should automatically create a new disk partition that fills the rest of the SD card (similar to how the Rasperry Pi OS expands to fill the SD card). The new disk partition should be where the RaSCSI disk images are stored.

As part of this effort, we should evaluate how much boot time is taken by services such as Samba, SFTP, Apache, etc. Ideally, the RaSCSI should be up and running within 2 seconds after power is applied.

Some things to consider.... we could start some of the services AFTER RaSCSI is started. The web server, file sharing service, SSH service, etc aren't needed immediately at power up.

akuker, 2020-08-24:
Additional resource:
https://www.furkantokac.com/rpi3-fast-boot-less-than-2-seconds/

akuker, 2020-08-24:
And another:
http://himeshp.blogspot.com/2018/08/fast-boot-with-raspberry-pi.html

akuker, 2021-03-10:
Some notes from "cag":

> As for faster spin up of the rascsi service…
> 
> I’m using a pi-zero-w and I’m connected over wireless.
> 
> Here’s the relevant info.
> 
> I followed the link on the github issue:
> 
> http://himeshp.blogspot.com/2018/08/fast-boot-with-raspberry-pi.html
> 
> Based on the steps in that article
> 
> Everything but the disabling of things (I did disabling via systemctl instead)
> 
> Verbatim
> 
> I used this command instead:
> 
> systemd-analyze plot > tree.svg
> 
> And disabled bluetooth, keyboard, triggerhappy, apt, (there may be more, I don’t know how to list the ones I disabled)
> 
> Instead of the basic.target, I changed the config for rascsi.service to only be dependent on the filesystem, I attached what I’m using.
> 
> Ignored
> 
> Ignored
> 
> Ignored
> 
> Verbatim
> 
> Ignored
> 
> My guess is that if you recompiled the kernel you could shave off another second or so based on the results he had from the article. I stopped tweaking once the system booted without the flashing disk, I expect someone with more knowledge could get it down further.
> 
> I know y’all are working on implementing network over scsi. One thing that seemed to take a long time to spin up was DHCP, I expect with hard coded info for the network even requiring networking could be a little faster.
> 
> Let me know if I can provide anything else to help.
>

akuker, 2021-09-05:
FYI…. Spent some time on this…. Making some progress on the buildroot_integration branch

akuker, 2021-11-18:
https://furkantokac.com/rpi3-fast-boot-less-than-2-seconds/

This is an example of a buildroot config that boots in less than 2 seconds!

philenotfound, 2022-01-09:
Do you need any help with this?

akuker, 2022-01-12:
> Do you need any help with this?

Absolutely!!

We've started going down a different route, building using the Raspberry Pi OS "pi-gen" framework. However, this still isn't as light weight as buildroot can be. So, there are a couple different ways to attack this:
 * Optimize the pi-gen configuration to speed up boot time as much as possible
 * Go back to buildroot

The pi-gen configuration is available here: https://github.com/akuker/rascsi-pi-gen
The buildroot work that I did is avialable here: https://github.com/akuker/RASCSI/tree/buildroot_integration

Note that the buildroot work has been on the shelf for quite some time. The branch is probably extremely out of date at this point.

uweseimet, 2022-08-26:
@philenotfound I am currentl browsing old tickets in ordert to clean up. I might be wrong, but you sounded as if you wanted to offer help. Any news on this, or maybe I just misunderstood?

akuker, 2022-09-01:
Meh... this ticket has been open forever. Closing it. We can revisit it when/if someone has time.

closed 2022-09-01 by akuker