PiSCSI/piscsi #27: Implement Macintosh-compatible Network Interface

akuker, 2020-08-26:
This issue is to request that an Ethernet network interface emulator is added to the RaSCSI software. 

The RaSCSI service currently has a "network tap" interface that allows the Sharp 68000 to use it as a network interface. This uses a custom driver, so its highly unlikely that this will work without significant MacOS driver development.

Saybur has done some awesome work with getting a microcontroller to work as a Nuvolink SCSI, which already has Mac drivers.
https://github.com/saybur/scuznet
https://68kmla.org/forums/index.php?app=forums&module=forums&controller=topic&id=58413
The easiest way to approach this might be to use his code as an example and implement similar functionality in RaSCSI.

There are probably other ways to do it as well. Discussion is welcome below...

cheesestraws, 2020-08-26:
More to the point, saybur has also _documented_ how to emulate being a Nuvolink SCSI and the protocol between the computer and device.  Which is great.  My bets would be to take that code and protocol notes (provided license is compatible, ofc) as a basis and this was the approach I was planning to take.

(Not being territorial here; if someone else gets to it before I do, great!)

akuker, 2020-09-09:
Another possible SCSI ethernet interface:
https://cdn.hackaday.io/files/18974811783616/SLINKCMD.TXT

akuker, 2020-09-23:
Note that the "USE_SEL_EVENT_ENABLE" functionality needs to be disabled from RaSCSI. The software will hang for 3 seconds whenever there is not activity on the SEL signal on the bus. This can delay the reception of a given packet for up to 3 seconds. 

The RaSCSI software seems to have been architected such that activity will only ever happen when requested from the host. There aren't any provisions for the target to do a "re-select" when it gets new data. 

The X68000 network tap was a polling driver. It was up to the X68000 to check if there were any packets periodically.

It might be possible to bring this back someday, but it should only poll for a few milliseconds.

akuker, 2021-01-30:
This change has been added to pull request #76.

The approach which used the Nuvolink was abandoned in favor of emulating a DaynaPort SCSI/Link. The Nuvolink required the "reselection" ability, which RaSCSI does not currently support. The DaynaPort SCSI/Link is periodically polled by the host, thereby not requiring the reselection logic.

Once the pull request has been closed, a new issue will be written to address the following remaining improvements:
- In scsi_daynaport.cpp, it uses a hard-coded MAC address. This should be updated to dynamically match the MAC address assigned by the Linux network stack.
- The multi-cast filtering isn't set up properly. Right now, it will only receive AppleTalk multi-cast messages. Any other multi-cast messages will be discarded. The way it *should* work is to listen for the 0x0D (eCmdSetMcastAddr) command, which will add which multicast address should be received.
- The ctapdriver should probably be updated to create the rascsi_bridge bridge if it doesn't already exist.

rdmark, 2022-01-09:
Closing this out since further improvement tickets have been created:
https://github.com/akuker/RASCSI/issues/598
https://github.com/akuker/RASCSI/issues/599
https://github.com/akuker/RASCSI/issues/600

@akuker This feature was a true killer app for RaSCSI. It's quite the feat! :)

closed 2022-01-09 by rdmark