PiSCSI/piscsi #28: Set up Coverity Analysis of the code

akuker, 2020-08-26:
Coverity static analysis software is an industry leading tool that is available for free to open source projects. It would help improve the quality of the code if we were to run this analysis on RaSCSI.

akuker, 2020-08-26:
I started setting up the project, but didn't get very far...
https://scan.coverity.com/projects/akuker-rascsi?tab=overview

uweseimet, 2022-02-07:
@akuker Any news on this? Not having automated builds of the develop and feature branches (in order to also have automated unit tests) and not having any code analysis, e.g. based on SonarQube, which is free for C++ for non-commercial projects (https://sonarcloud.io/explore/projects) is quite a problem. The more code is added without automated testing, the more likely regressions get.

akuker, 2022-02-13:
Agreed. Its all a matter of priority. This isn't an especially "sexy" feature to work on, so its easy to push it down the priority list. I definitely have not forgotten about it.

uweseimet, 2022-02-13:
@akuker Yes, just like documentation tickets like this are not that popular. But the more code we have, the more important this ticket gets.
My experience from previous projects is that having code analysis and working steadily on improving code metrics helps preventing a lot of issues. And it uncovers a lot of issues, which would otherwise have taken a lot of time in debugging the problems they would have caused.

rdmark, 2022-08-27:
@akuker What did you get stuck on setting up the Coverity project? I can see that you got it to scan a whole bunch of code, so you must have made some progress!

uweseimet, 2022-08-31:
@akuker @rdmark I think I have resolved this ticket, please see https://github.com/akuker/RASCSI/pull/806.

rdmark, 2022-08-31:
Coverity is very similar to SonarQube in its capabilities, I believe.

akuker, 2022-09-01:
Coverity looks like it is similar to SonarQube (from my 30 second google search of SonarQube). Coverity can identify dead code, but that's not its primary function. Its more like lint on steroids.

I'm excited to see gcov being used. As for this ticket, I don't care. We can close it, we can leave it open. Whatever.

There are other priorities... I'll just close it.

closed 2022-09-01 by akuker