PiSCSI/piscsi #37: Allow dynamically attaching a user defined set of drives dependant on computer attached

erichelgeson, 2020-09-16:
For example:

attach a Macintosh Plus would mount
/home/pi/images/MacPlus6.0.8.hda
/home/pi/images/OS6Software.hda
etc..

attach a Macintosh SE/30 would mount
/home/pi/images/MacSE30-7.5.5.hda
/home/pi/images/OS7Software.hda
etc..

attach from a Macintosh LC would mount
/home/pi/images/MacLC-7.5.5.hda
/home/pi/images/OS7Software.hda

If we can capture what type of computer is asking for devices (id, serial#, some other unique id) and then mount them dynamically based on a config file it would be quite awesome for us who have many computers - we wouldn't need to ssh/web into RaSCSI to re-configure.

kraptv, 2020-11-20:
This is pretty hard to do in practice.  I understand this is not easy because no host will accept any kind of SCSI Mode Sense from its devices, only the host can query this of its devices. 

You MIGHT be able to detect some unique behaviors based upon mapping out the quirks of each machine's SCSIGet behavior quirks, but I don't think there is enough variance between the Macs to be able to identify them as such. (Many Macs use the 53C96 and 53C94 chips and would be abstracting their behaviors behind this)

https://developer.apple.com/library/archive/documentation/mac/pdf/Devices/SCSI_Manager.pdf

uweseimet, 2021-08-09:
Note that more than one computer can be attached to the same bus, provided that each computer (initiator) has a different SCSI ID. I have already worked with setups like this. As long as both computer support SCSI bus arbitration, SCSI access conflicts are avoided. (Of course, this does not resolve issues caused by the computers independently caching disk data.)
WIth more than one computer on the bus no decision whatsoever can be taken on the set of drives to be used.

uweseimet, 2021-09-02:
@akuker I suggeset to close this ticket, because any computer can send any SCSI command or sequence of SCSI commands. It's impossible to identify a computer based on a SCSI command received. SCSI itself is agnostic of the computer sending the command.

akuker, 2021-09-03:
I agree that this request would be extremely difficult to do and of limited value. The risk of breaking something is too high (IMO)

closed 2021-09-03 by akuker