PiSCSI/piscsi #40: Mac Plus will not boot from RaSCSI

akuker, 2020-09-30:


erichelgeson, 2020-09-30:
- Which version of Pi: 4b 4gb
- Github revision of software: cb295b89b95d1c38b0cd614d49b2e3f024ac9d73
- Board version: 2.2 (first tindie batch)

Booted via FloppyEMU saw the HDD fine and was able to read/write.

Used system picker to select the 6.0.8 of my bootstrap image (see discord) 

Rebooted - saw happy mac for a second, then ? (that did not flash)

kraptv, 2020-11-20:
Could this be related to the Mac Plus not being able to handle Unit Attention signals?

https://macgui.com/news/article.php?t=471
https://www.codesrc.com/mediawiki/index.php/Mac_Plus
https://spinsidemacintosh.neocities.org/tn405.html#tn096

rdmark, 2021-09-15:
It would be great if someone with a Mac Plus could retest this on the latest develop, and perhaps provide some trace logs if it still is a problem!

Aapjedotinfo, 2021-09-17:
I've just booted my Macintosh Plus (with 4MB) with an image of 6.0.8. Believe im running version 21.05. Does that help?

rdmark, 2021-09-17:
@Aapjedotinfo Thank you for testing! You're running the latest official release.  May I ask how you created the disk image you booted off here?  Did you use the easyinstall.sh script to generate a Mac image file, or did you create a blank image from scratch and formatted it from within the Mac System Software (e.g. using Apple HD SC Setup)?

The reason for asking is speaking with @erichelgeson he suggests the original issue was caused by the Lido driver that easyinstall.sh uses for Mac drive image formatting.

Aapjedotinfo, 2021-09-17:
I've used a image i had before that also works in SheepSaver. Got it from a 'guy' ;-) ( source: https://www.savagetaylor.com/downloads/downloads-macintosh/ )

As a newbie with this new device, i had no luck with the 'empty' image feature. But i will give it a go, just switched to the develop branch tho.

rdmark, 2021-09-17:
@Aapjedotinfo Thanks for the link, I'll play around a bit with these images myself.

With regards to empty images, my preferred approach is to create a generic 'hda' image (Apple Genuine) file and then format it using the [patched Apple HD SC Setup utility ](https://macintoshgarden.org/apps/apple-hd-sc-setup-753-patch)that bypasses the drive authenticity check.

BTW, the develop branch is under active development right now so please by all means give us feedback, feature requests, or bug reports. I guarantee it will be looked at quickly.

Aapjedotinfo, 2021-09-18:
Had a bit more issues with my Macintosh Plus when i plugged in my RaSCSI... It goes into some weird blackscreen with grablenoise on the speaker. Had it happen on two of my Plus machines now. Cannot really tell what the issue was, looked to be resolved after a reboot of the Pi. But currently i cannot reproduce it anymore. 

But i have a feeling the errors i had, might have something to do the issues scsi2sd has too. As @kraptv mentioned the following url mentions something about the bootrom quiting after 1ms instead of the standard of 250ms. The garble sound that my Mac made was a bit 'worrying' (sounded like it was about to explode). Are these settings anything RaSCSI can set?
https://www.codesrc.com/mediawiki/index.php/Mac_Plus 

Anyway, i tried the empty image now with the patched utility. I had no issues with formatting it and installing 6.0.8 on it via diskettes. 

Also i dont know if the starter of the issue had the termination switches correct on their rascsi? But i hope that is too obivious :-)

rdmark, 2021-09-18:
@Aapjedotinfo I had the exact same thing happen with RaSCSI on my Mac SE once (2.4a board), with that loud noise coming out of the speaker and screen freezing up.  I *think* this is caused by the Pi itself overheating and CPU slowing down, which leads to the RaSCSI software not being able to keep executing the SCSI emulation fast enough for the host system.  I don't have evidence to prove this hypothesis, but the circumstances points towards this.

@akuker Do you have any thoughts on the above?

Aapjedotinfo, 2021-09-18:
@rdmark I'd have to say, my Pi4 is very hot now you mention it. Also the RASCSI board IC's are pretty hot to the touch, therefor i've put those little stick-on heatsinks on them... I've got no clue if thats normal/by design (fullspec board, 2.4a aswell).

But when it was crashing my Plus i've tried my Classic II, that had no issues. So i doubt it was the Pi....

rdmark, 2021-09-18:
@Aapjedotinfo I haven't seen the issue happen on my Performa 476 either, so the common factor here could be the 8Mhz CPU. Anyhow, we're getting pretty far away from the original issue of booting the Plus.  Might I ask you to create a new issue ticket for the crashing issue?

erichelgeson, 2021-09-18:
The weird freezing/popping/etc happens when using the LIDO SCSI drivers. If you use a different image with Apple's drivers it does not happen. Had the same issue on BlueSCSI on a Plus when using LIDO.

rdmark, 2021-09-18:
I added a small tweak to the easyinstall.sh script to make it clear that there are some caveats to using the Lido driver. In my PR https://github.com/akuker/RASCSI/pull/242

rdmark, 2021-09-18:
Let me close this ticket out at this point. Please report new tickets for other Mac Plus issues!

closed 2021-09-18 by rdmark

baragona, 2022-04-27:
Since this can work, should the red X on this page https://github.com/akuker/RASCSI/wiki/Compatibility be changed to a green check?

joeventura1, 2022-07-11:
Since Lido causes all this, what is the best tool to init a SCSI volume on a RaScsi? (Mac Plus) The apple tool, even patched does not even see it.

rdmark, 2022-08-19:
@joeventura1 At the present moment, Silverlining seems to be the best choice. See https://github.com/akuker/RASCSI/issues/656