PiSCSI/piscsi #43: When running a `rasctl` command that fails, the command should exit with a non-0 exit code

erichelgeson, 2020-10-12:
# Info

- Which version of Pi are you using: 4b 4gb
- Which github revision of software: `cb295b8`
- Which board version: 2.2
- Which computer is the RaSCSI connected to: N/A

# Describe the issue

When running a `rasctl` command that fails, the command should exit with a non-0 exit code

```
$ rasctl -i 3 -c attach -t hd -f /home/pi/images/test.hda ; echo $?
Error : File open error [/home/pi/images/test.hda]
0 # expect this to be non-0 as it failed.
```

uweseimet, 2021-07-08:
The main issue IMO is the primitive interface between rasctl and the rascsi process. Commands to be processed by rascsi can only return a string, which is usually used for logging. A clean solution would require to return structured data.
@akuker protobuf (https://developers.google.com/protocol-buffers) looks like an option, doesn't it?

akuker, 2021-07-08:
@uweseimet - I had played with improving things a little bit a while back https://github.com/akuker/RASCSI/tree/cmd_interface_update/src/raspberrypi but never got back to it. 

Protobufs looks like a lot better option though.

uweseimet, 2021-07-19:
Resolved by https://github.com/akuker/RASCSI/pull/129.

```
rasctl -i 3 -c attach -t hd -f /home/pi/images/test.hda ; echo $?
Error : File open error [/home/pi/images/test.hda]
107
```

closed 2021-07-19 by uweseimet