PiSCSI/piscsi #48: Equate benchmarks with Mb / MB / s

dapperrobotbear, 2020-10-27:
Norton system info doesn't really give us performance in terms that are relatable to other hardware solutions or original bus speed maximums. It would be nice to know what the RASCSI is truly capable of.

erichelgeson, 2020-10-27:
The detailed reports in Norton System Info do give the detailed numbers - https://github.com/akuker/RASCSI/wiki/Benchmarks#disk-read-scores

Is there another utility you prefer or someone could run and add to the wiki?

dapperrobotbear, 2020-10-27:
Sorry, am I misreading the information? It gives scores on all those columns but that's not exactly an indicator of speed, only comparative "score". Is that supposed to indicate the number of times it could run that test in a fixed time? There's nothing to equate the numbers with. Linux has a transfer speed utility but I'm not sure what would be available to OS8-9. https://www.cyberciti.biz/tips/how-fast-is-linux-sata-hard-disk.html

I only ask because I'm wondering if RASCSI can achieve speeds at or above Wide SCSI-2. Which is around 5MB/s.

erichelgeson, 2020-10-28:
Ahh your right in those screen shots - IIRC one of those drop down would show Xb/sec. Maybe someone with a scsi card on linux could run those speed tests.

kraptv, 2020-11-20:
Generally networking is in terms of bits ( "I have gigabit internet connectivity") and storage throughput is in terms of bytes and IOPS. ("My SATA HDD can do 90 Megabyte/sec at the edge and 50 Megabyte/sec towards the spindle")

Please do not use little b (Megabit, Gigabit) nomenclature, use big B nomenclature (Megabyte, Gigabyte, Terabyte) when speaking about data persistence and transfer. 

One could run iozone3 under Linux or iometer under Windows if someone wanted to be consistent.  However, please note that the SD cards on the Raspberry Pi should be of sufficient speed (Use a UHS-3 / A1 / A2 rated card, not some freebie cheapo Class4/Class10 card please) to eliminate bottlenecking elsewhere.

uweseimet, 2022-08-26:
@akuker I also wondered about the actual performance in the past, but I don't think I ever got a response when I asked about the maximum throughput possible with the existing hardware ;-). If these data cannot be provided I suggest to close this ticket.

akuker, 2022-09-01:
Agree. Closing ticket.

closed 2022-09-01 by akuker