PiSCSI/piscsi #49: This should be a general-purpose repo and credit the author in readme.md.

ianhan, 2020-10-27:
I can't seem to find anything in the changelog that add to the original functionality implemented by GIMONS. He is not mentioned in the readme.md, and this project is Mac-focused for some reason, but there aren't any Mac-specific features to RaSCSI. The FullSPEC board design is nice.

I am currently looking at using RaSCSI with NeXT machines and SCSI-capable retro PCs.  Could this repo be repurposed as a general RaSCSI repo (after all it is just a SCSI emulator), mention GIMMONS as the creator in the readme.md, and be a place for updates to the Pi software in general?

rdmark, 2021-08-20:
I assume things have changed since this issue was created, since as of the time of writing I believe these concerns has been resolved?
There is nothing in the README.md that suggests this is a Mac specific project, and points to the wiki landing page https://github.com/akuker/RASCSI/wiki where GIMONS is clearly called out as the original creator. 
Some documentation points to setup instructions for the Atari ST platform, and the man pages have clear references to the original Sharp X68k support.
Is there anything else to address here that I'm overlooking?

rdmark, 2021-08-20:
Ok, looking closer I see two potential areas of improvement:

About in the right rail says:
> English version of the Raspberry Pi SCSI interface board "RaSCSI". This allows a Raspberry Pi to function as an emulated hard disk and/or CD-ROM for vintage Macintosh computers. 

The readme has this sentence:
> This project is aimed at users of vintage Macintosh computers from the 1980's and 1990's.

I agree that even if the overall project wasn't officially pivoted to a more general-purpose solution, it'd be more accurate to qualify both of those statements with "...and other vintage computers that use the SCSI bus" or something like this.

akuker, 2021-09-03:
I agree with the statements. When I first created this fork, the target was Macs. This has grown FAR beyond Macs, and this should be cleaned up someday. 

I also want to rename the project from “RaSCSI - 68kmla version” to something like “RaSCSI Reloaded” or something.

rdmark, 2021-09-06:
@akuker are you brainstorming suggestions? :)
How about:
- RaSCSI Redux
- RaSCSI Mk II
- RemasterSCSI

akuker, 2021-09-25:
I am open to suggestions! RaSCSI Reloaded just seems to resonate with me for some reason :-)

rdmark, 2021-11-04:
The top level README.md as well as the right rail blurb has been updated now with the release of version 21.10, so I consider the original issue resolved. It should be clear for newcomers that RaSCSI is suitable for any SCSI enabled device.

closed 2021-11-04 by rdmark