PiSCSI/piscsi #54: Formatting with Apple HD SC Setup only makes 20mb drive

erichelgeson, 2020-11-11:
# Info
Create a 100mb drive, mount it, attempt to format with Apple HD SC Setup 

Expected: 100mb drive 

Actual: 20mb drive 

- Which version of Pi are you using: 4b
- Which github revision of software: latest 
- Which board version: 2.2
- Which computer is the RaSCSI connected to: Classic 1

# Describe the issue

erichelgeson, 2020-11-11:
Likely cause:

> I came across an interesting issue when playing around with PCE/macplus. I used Linux dd command to create a 1000MB hard disk image, attach it to the machine using the config file, and use Apple HD SC Setup to initialize the drive and create the partitions. Surprisingly, despite changing various options, the tool only created a 20MB partition while leaving the rest of the emulated hard drive unallocated. I have been using this tool on several large hard drives up to 2GB, both emulated and real, with no issues. So what was wrong this time?

> After much debugging, the issue is due to PCE/macplus faithful attempt to emulate the original Apple hard disk drive, down to specifying SEAGATE and ST225N as the vendor and product strings in the config file respectively:
>device {
>   id      = 0
>   drive   = 0x81
> 
>   vendor  = " SEAGATE"
>   product = "          ST225N"
>}

>This config emulates the Seagate ST225N 20MB hard disk drive that comes with many compact Macintoshes of the time. Although this is good in a sense because the emulated hard drive will show up in Apple HD SC Setup immediately without the need for a patch, it also causes Apple HD SC Setup to think that the hard drive is only 20MB and creates a single 20MB partition, without even querying the drive for its real size. The solution is to change the vendor and product name to something else (while maintaining the original string length), and use the patched version of Apple HD SC setup to initialize the drive.

http://www.toughdev.com/content/2016/11/pcemacplus-the-ultimate-68k-classic-macintosh-emulator/

sonique6784, 2020-11-13:
With the new `easyinstall.sh` assistant, we can create Hard Drives that are already formatted HFS.  This isn't solving this particular problem, but at least it can help some people.
When I get the opportunity, I will link up the Web UI with `easyinstall.sh`, so that we can create drives easily.

BleuLlama, 2021-03-21:
I can confirm this also happens on my system. Pi 3b, develop branch, v2.3 board, connected to Mac Classic II.
with the HD SC setup for 7.5.3.

rdmark, 2021-09-15:
@erichelgeson @BleuLlama This should be resolved now in the latest develop (8a3642b) with RaSCSI reporting the following SCSI inquiry information by default when a 100MB image is attached:

- Vendor: RaSCSI
- Product: SCSI HD 100 MB

I tested using [Apple HD SC Setup 7.5.3 ](https://macintoshgarden.org/apps/apple-hd-sc-setup-753-patch) (patched) and it automatically identified and initialized a file system with 98.2 MB available to the system.

Of course, if you configure RaSCSI to present different inquiry information that Apple HD SC Setup will identify as, say, a 20 MB Seagate drive, then it is likely to format only a 20 MB partition even if you present it with a larger image. E.g.:

> rasctl -i 3 -c attach -f ~/images/testseagate.hda -n "SEAGATE:ST225N: "

rdmark, 2021-09-28:
This is resolved in https://github.com/akuker/RASCSI/pull/275 by spoofing as QUANTUM:FIREBALL for hda

closed 2021-09-28 by rdmark