PiSCSI/piscsi #55: Web Interface suggestions: Deny ID 7 (or warn) on Mac-centric SCSI assignments

kraptv, 2020-11-20:
In the RaSCSI web interface, ID 7 seems to be treated like a normal ID that can be attached, but we all know that it is the Mac host itself that reserves ID7 and would cause trouble.  I am not so worried about the command line as most people will be at least a little more lucid and aware that the host is 7, but you may want to call this out on the web interface.

Consider hard-coding a requirement of 'are you sure you know what you're doing' or just make SCSI ID 7 invisible for users of the web management.

ewanuno, 2020-11-26:
Akai samplers default to ID6 so it might be better to stick with a warning.
a simple exclamation mark with a tooltip would be fine by me.

it is possible to list scsi devices already present on the bus so it should also be possible to prevent loading images on active scsi ids

kraptv, 2020-11-26:
SGI systems use ID 0, so it’s a slippery slope here. Given the target user profile at the moment, maybe a warning after selection could be a better route?  Selecting ID 0 when it’s usually allocated for the Mac’s internal drive could be a thing too.

uweseimet, 2021-08-09:
Considering that with some platforms (e.g. Atari ST) ID 7 is legal I doubt that there can be any Mac-centric solution. How would RaSCSI know when or when not to display the warning?

rdmark, 2021-08-30:
There is already implementation for this in rascsi-web -- in 62f6c8d develop on line 11-12 of ractl_cmds.py:
`# List of SCSI ID's you'd like to exclude - eg if you are on a Mac, the System is usually 7`
`EXCLUDE_SCSI_IDS = [7]`

This disallows assignment on the specified ids through the web ui.

Line 56 of 'templates/index.html' also controls which ID will be excluded from the list of devices in the web ui.

It would still be nice to have some more graceful configuration options here for non-Macs, but at least there's some logic in place.

rdmark, 2021-08-30:
Created https://github.com/akuker/RASCSI/issues/220 for looking into making the host id configurable.

closed 2021-09-15 by rdmark