PiSCSI/piscsi #56: Web interface suggestion:  Disallow attaching the same image to be attached to multiple SCSI IDs.

kraptv, 2020-11-20:
- Which version of Pi are you using: Raspberry Pi 4 4GB Model B
- Which github revision of software: 524f1c7826f8353439cf6e3f30d6f5b8f4581648
- Which board version: 2.3 FULLSPEC
- Which computer is the RaSCSI connected to: Quadra 700 and PowerMac 8500

I saw that when I've attached an image via the web browser, it remains available for attaching on others and can do this freely.  Consider iterating through the 'already attached' image array before populating the 'available images' to attach web menu. 

Having the same disk volume presented as multiple disks can produce erratic results.

uweseimet, 2021-07-20:
A set/map with image names/paths should be introduced, so that rascsi can report an error in such a case.

uweseimet, 2021-08-06:
With https://github.com/akuker/RASCSI/pull/162 rascsi rejects attaching the same image more than once.

closed 2021-09-15 by rdmark