PiSCSI/piscsi #57: Web interface suggestion: thin-provision disk images (and consider revealing bytes written)

kraptv, 2020-11-20:
# Info

- Which version of Pi are you using: Raspberry Pi 4 4GB Model B
- Which github revision of software: 524f1c7
- Which board version: 2.3 FULLSPEC
- Which computer is the RaSCSI connected to: Quadra 700 and PowerMac 8500

# Describe the issue
Create disk images using sparse files and don't make the user wait for worthless zeroes being written to disk.

Sparse file disk images work great with RaSCSI!  I would absolutely recommend this as you can see how much disk space is actually used (after formatting!) on the RasPi side. Note that this is only at time of first write and not if you fill the thing up and then delete a bunch of stuff.

Example is creating a 2GB disk image but populating it sparesely on the command line - this translates easily into a tweak in PHP code on the web interface side. 

dd if=/dev/zero of=sparse_drive.hda bs=1 count=0 seek=2000M
(then format disk  using Lido or similar from inside the system and copy over some crap)
then see the "official reporting" using this command:
pi@raspberrypi:~/images $ ls -lh sparse_drive.hda 
-rw-r--r-- 1 pi pi 2.0G Nov 14 20:02 sparse_drive.hda
and now show the 'effective disk space using the following command':
pi@raspberrypi:~/images $ du -sh sparse_drive.hda 
15M     sparse_drive.hda

Think of it as "thin provisioning" your MacOS disk images.

Note that copying a thinly provisioned file to another file system will reinflate all of those zeroes, etc. etc. but for those that are just using a RaSCSI as an external drive and not copying data around outside of the Linux filesystem, it is pretty neat.

rdmark, 2021-09-18:
@uweseimet With the changes you're making in https://github.com/akuker/RASCSI/pull/240 we're moving to the backend side handling image file creation.  Can you comment on whether it's feasible to create 'sparse' image files with rascsi?  It seems like a fairly useful feature.

uweseimet, 2021-09-18:
@rdmark As far as I know sparse files are fine for binary files that are written once and then only read, but are not recommended for files which are modified, like images files. In addition, the files create by fallocate() usually do not have a lot of zeros in them, because they are not explicitly filled with content when creating them.
So all in all, I don't think that sparse files should be supported.

rdmark, 2021-09-18:
Thanks for the analysis. Let me close this ticket.

closed 2021-09-18 by rdmark