PiSCSI/piscsi #58: Make CDROM disk attach errors more descriptive

kraptv, 2020-11-20:
# Info

- Which version of Pi are you using: Raspberry Pi 4 4GB Model B
- Which github revision of software: 524f1c7
- Which board version: 2.3 FULLSPEC
- Which computer is the RaSCSI connected to: Quadra 700 and PowerMac 8500

This behavior is consistent via the web interface or command line.

When attaching a disk image (or anything strange that doesn't resemble a disk) as a CDROM, the error "File open error" is ambiguous and implies that it's possibly an issue with the underlying file system (permissions, locking, etc) rather than the ability to parse the file format of the accessed image.  

**Error message: Error : File open error [/home/pi/images/cdstuff1995.img]**

Consider changing "File open error" to "CD format unreadable" or similar.  Consider a warning that states you can mount raw/ISO style CD images but not DiskCopy disk images (dmg/img format would be a WONDERFUL FEATURE THOUGH - Requires a little bit of hacking though!)

uweseimet, 2021-08-09:
@kraptv Please check whether PR https://github.com/akuker/RASCSI/pull/175 resolves this issue.

rdmark, 2021-08-20:
I can confirm that where you in master get a generic error when mounting an invalid CD image, I now get this for the same CD images:
> Error: Tried to open an invalid file '/home/pi/images/71A.iso': ISO CD-ROM file size must be a multiple of 2048 bytes 

uweseimet_develop b32f324