PiSCSI/piscsi #59: Web interface suggestion: Provide option to persist disk images and their SCSI IDs over reboot

kraptv, 2020-11-20:
# Info

- Which version of Pi are you using: Raspberry Pi 4 4GB Model B
- Which github revision of software: 524f1c7
- Which board version: 2.3 FULLSPEC
- Which computer is the RaSCSI connected to: Quadra 700 and PowerMac 8500

# Describe the issue

Typically, we setup our RaSCSI to connect to the computer, select our boot drives, and get everything configured nicely via web interface and then forget about how we setup the system.  Then the power fails and we lose the whole config again.  

It'd be very nice to either have the image-to-SCSI ID have an optional 'persist upon reboot' assignment at time of attachment or "Save config" button on the main page that the service will read and configure upon reboot.

kraptv, 2020-11-20:
Alternately, it'd be nice to have a rascsi.conf file where these things would persist, potentially with overrides (vendor/model/serial-number examples that enable certain formatters, for example)

rdmark, 2021-08-19:
For the record, the way that you make device configurations persistent through reboots is to edit the rascsi.service configuration file typically residing in /etc/systemd/system/

I still agree that having the convenience of editing this through the webui would be great.

erichelgeson, 2021-08-25:
This exists in develop - you can save pre-sets and if it's name default it will load when rascsi-web loads.

rdmark, 2021-08-30:
I couldn't find a trace of the original implementation, so I went ahead and coded a new one in https://github.com/akuker/RASCSI/issues/217

@kraptv Can you please confirm that this resolves your intended usecases?

closed 2021-09-15 by rdmark