PiSCSI/piscsi #60: Web interface vs. command-line inconsistency: disallowing .dsk .dmg .img .cd .toast images

kraptv, 2020-11-20:
# Info

- Which version of Pi are you using: Raspberry Pi 4 4GB Model B
- Which github revision of software: 524f1c7
- Which board version: 2.3 FULLSPEC
- Which computer is the RaSCSI connected to: Quadra 700 and PowerMac 8500

# Describe the issue

The web interface (rightfully) refuses to upload anything other than .hda or .iso images, however, if any other 'emulated mac' disk files (.dmg, .img, .dsk, .image, .toast) happen to be copied into the images directory, the web interface will attach the disk images fine.  However, some (many? all?) of these disk images will fail to be recognized by the Finder operating system or any disk repair utility due to weird looking partition tables presented.

rdmark, 2021-08-19:
The filename extension of a CD image file does not accurately reflect its ability to be mounted in MacOS, by my knowledge and experience.  It has more to do if the metadata and size meets the standards expected by RaSCSI, the supported file system and drivers active.  I've successfully mounted images with .toast and .img file endings, in addition to .iso.  See detailed test report here:  http://shorturl.at/wKNY9

rdmark, 2021-09-15:
In addition to the above point, RaSCSI use is not limited to the Mac, but can be used on at least a dozen other computing platforms. What RaSCSI does it to present the image to the host system, and if it cannot read the partition table / file system then it will attempt to initialize it. Hence, any file with the right block size (i.e. a multiple of say 512 bytes) can be a valid disk image.

As per this discussion, please let me close this ticket. I suggest you try out the latest develop and let us know any other improvement suggestions based on how it functions now.

closed 2021-09-15 by rdmark