PiSCSI/piscsi #61: Restrict image upload on filetype attribute instead of frustrating user AFTER they have uploaded.

kraptv, 2020-11-24:
# Info
- Which version of Pi are you using: Raspberry Pi 4 4GB Model B
- Which github revision of software: 524f1c7
- Which board version: 2.3 FULLSPEC
- Which computer is the RaSCSI connected to: Quadra 700 and PowerMac 8500

# Describe the issue

If you want to restrict the filetype of the images being uploaded, please use the accept attribute of input type = "file" so that file selection dialog boxes won't allow it. 

Something along the lines of:

<input type="file" accept=".iso,.hda" />

It is a bad UX to wait on an upload of an image and then get the womp-womp "sorry, not possible" note at the end.  Alternately, you could suggest to rename it as frequently .toast images can endure a simple suffix rename and be perfectly fine with mounting.

rdmark, 2021-08-30:
As of develop 62f6c8d rascsi is actually able to attach a broad range of image files, not restricted to iso or hda line endings.  For instance, img, cdr, or toast files are interpreted correctly and mounted as CD-ROM images now (if they are in a valid format otherwise.)

In general, I don't think it's a good idea to define a finite allow-list for file types for files that you use with RaSCSI, since it's deeply flexible at its core and potentially can be used with a with a broad range of media such as magneto-optical or zip disks for a dozen different computer platforms that support SCSI.

rdmark, 2021-09-24:
@kraptv This has been resolved in the latest develop code now. Please try again when you have the chance.

closed 2021-09-24 by rdmark