PiSCSI/piscsi #63: Web interface: feature request: cdrom metadata files

ewanuno, 2020-11-26:
cdrom images often have important info on the cover, or even associated manuals
it would be very useful to have these files displayed in the web interface alongside the .iso and available for viewing directly in the browser.

for ease of implementation, i propose that a cdrom image lives in a folder with its associated images and text/pdf files
which all have the same name as the image file.

e.g. : 
SuperMacintosh_multimediaCD_cdimage/

SuperMacintosh_multimediaCD.iso
SuperMacintosh_multimediaCD_frontcover.png
SuperMacintosh_multimediaCD_backcover.png
SuperMacintosh_multimediaCD.pdf
SuperMacintosh_multimediaCD.nfo
SuperMacintosh_multimediaCD.txt

rdmark, 2021-10-17:
@ewanuno Thanks for raising the improvement idea! I've been working on expanding the functionality of the Web UI, and want to consider the use case here.

The design philosophy behind the Web UI is something light-weight and HTML4 backwards-compatible that functions without unnecessary reliance on JavaScript (and compatible with Netscape 4). As such, I would strongly advise against introducing embedded images, since 68k Macs for instance will choke on them.

Is there any other kind of metadata beyond cover images and pdf files that would be helpful here?  Textual metadata in nfo files?

rdmark, 2022-08-19:
This is not something we will add to the default Web UI, since the design philosophy is a very light weight web app that can run on any vintage computer/browser. Adding inline images is going to make it too heavy on very old systems.

closed 2022-08-19 by rdmark