PiSCSI/piscsi #64: Footprint values on 2p3 are TH instead of SMD.

StormTrooper, 2020-11-29:
Using the latest master, if I open the PCB layout file the resistor footprints are all correct 0402, however opening the schematic the footprint values are still TH:

Resistor_THT:R_Axial_DIN0204_L3.6mm_D1.6mm_P1.90mm_Vertical

This causes an issue if you update the PCB layout and pull in the latest footprints. 

One other small issue on the J8 reference. Its used on both sides of the PCB. On the bottom side it caused an overlap with TP1,TP3

[image]

akuker, 2021-01-02:
Not sure how that happened - that's weird. I've corrected it in the "2.4" version that I'm working on. Will investigate this further sometime soon...

akuker, 2021-01-22:
I've fixed the resistor issue in version 2.3. The silkscreen issue will be fixed in the 2.4 version.

Reference commit d84b09595a0a597a8be0c4b7af66ae546d6af576

closed 2021-01-22 by akuker