PiSCSI/piscsi #72: Issue creating/mounting small disk images under 10MB

TomRBarber, 2021-01-17:
# Info

- Which version of Pi are you using: Zero W
- Which github revision of software: Latest build of the Daynaport branch
- Which board version: 2.3
- Which computer is the RaSCSI connected to: Apple IIe / Sandwich II SCSI card

# Describe the issue

It doesn't seem to like to mount small disk images under 10MB.  Both through CLI and web interface

Web interface says it mounted, but doesn't actually mount

CLI says "Error : File open error [/home/pi/images/new_drive2.hda]"

**Works -**
pi@raspberrypi:~/images $ dd if=/dev/zero of=/home/pi/images/new_drive8.hda bs=512k count=20
20+0 records in
20+0 records out
10485760 bytes (10 MB, 10 MiB) copied, 0.517814 s, 20.3 MB/s
pi@raspberrypi:~/images $ rasctl -i 3 -c attach -f /home/pi/images/new_drive8.hda

**Doesn't work -**
pi@raspberrypi:~/images $ dd if=/dev/zero of=/home/pi/images/new_drive7.hda bs=512k count=19
19+0 records in
19+0 records out
9961472 bytes (10 MB, 9.5 MiB) copied, 0.490213 s, 20.3 MB/s
pi@raspberrypi:~/images $ rasctl -i 3 -c attach -f /home/pi/images/new_drive7.hda
Error : File open error [/home/pi/images/new_drive7.hda]

erichelgeson, 2021-01-17:
Might be related to https://github.com/akuker/RASCSI/issues/70

RE: web not showing an error is dependent on https://github.com/akuker/RASCSI/issues/43

uweseimet, 2021-07-20:
@akuker I think this issue can be closed.

akuker, 2021-07-21:
Agreed!

closed 2021-07-21 by akuker