PiSCSI/piscsi #82: The tale of RaSCSI and SCSI RESERVE

phrax0, 2021-03-04:
# Info
- Which version of Pi are you using:
Raspberry Pi 3 Model B a02082 (Sony, UK)

- Which github revision of software:
Current, pulled down 2021-03-03

- Which board version:
FULLSPEC

- Which computer is the RaSCSI connected to:
Sun SPARCstation 5

# Describe the issue
Upon presenting a 120MB disk to a Sun Sparcstation, Solaris can see the disk without issue. 

```
c0t5d0          Soft Errors: 0 Hard Errors: 0 Transport Errors: 0
Vendor:  SEAGATE Product:           ST225N Revision: 0200 Serial No:
Size: 0.13GB <125644800 bytes>
Media Error: 0 Device Not Ready: 0 No Device: 0 Recoverable: 0
Illegal Request: 0 Predictive Failure Analysis: 0
```

When trying to format the disk with the format tool in Solaris, it fails with "Reserve failed".

```
format> current
Current Disk = c0t5d0
<SEAGATE-ST225N-0200 cyl 1226 alt 2 hd 8 sec 25>
/iommu@0,10000000/sbus@0,10001000/espdma@5,8400000/esp@5,8800000/sd@5,0

format> format
Mode sense page(4) reports rpm value as 0, adjusting it to 3600
Ready to format.  Formatting cannot be interrupted
and takes 14 minutes (estimated). Continue? yes
Beginning format. The current time is Thu Feb 25 22:34:42 2021

Reserve failed
failed
format>
```

It looks like rascsi does not respond to the RESERVE command.

17.2.8 RESERVE command
The RESERVE command (see table 343) is used to reserve units or, if the element reservation option is implemented, elements within units for the use of the initiator. The third-party reservation allows units or elements to be reserved for another specified SCSI device. The RESERVE and RELEASE commands provide the basic mechanism for contention resolution in multiple-initiator systems.

https://www.staff.uni-mainz.de/tacke/scsi/SCSI2-17.html

I don't think Sun systems are the target audience for RASCSI, so this is a low priority issue.

Thank you!

akuker, 2021-03-05:
> This command has been declared obsolete by the T10 committee. However, it is included because it may be implemented on some products.

Pretty much all of the computers we're using with RaSCSI are obsolete, so it seems like we should support this command :-)

From my quick research, this command looks like its used when there are multiple initiators on a SCSI bus. [Additional Background](https://docs.oracle.com/cd/E19050-01/sun.cluster31/817-6537/x-17eg5/index.html)

Doing the full RESERVE/RELEASE function could be a non-trivial amount of work, and probably won't be used by anyone. So, for now, I propose that these commands will just be implemented where RaSCSI will just respond with "success" whenever a RESERVE or RELEASE command is received. 

I'd recommend implementing RESERVE(6), RESERVE(10) and RELEASE(6), RELEASE(10), since they work the same way. 
https://origin-www.seagate.com/files/staticfiles/support/docs/manual/Interface%20manuals/100293068j.pdf

phrax0, 2021-03-07:
Thank you for the suggestion.  I'm working on those changes right now.

turbolent, 2021-06-16:
@phrax0 I see that in https://github.com/akuker/RASCSI/commit/1fb0269645feda44d689a597339bef332efd9258#diff-50ff296f47000fe3386f4f6f6c81b58d784ef16b31ebb21833d0284cf5f14d09 you stubbed out the reserve and release calls. 

Did you manage to re-test on your SPARCstation with these changes?

uweseimet, 2021-08-09:
@phrax0 Any news on this?

akuker, 2021-08-10:
Last I heard from @phrax0 , this was working. He's been AFK for a while, so we should probably leave this open until someone with a SPARCstation confirms.

nsafran1217, 2021-09-18:
I believe this issue is fixed. I just did a clean install of Solaris 2.6 onto a 2GB disk on the RaSCSI on a SPARCStation 5 with no issue. I also formatted a 200MB disk in Solaris after the install and it worked.

rdmark, 2021-09-18:
@nsafran1217 Which version of the RaSCSI software did you test with?  The master and develop branches have diverged a huge amount, and I'm curious to know if it works on develop in particular!

nsafran1217, 2021-09-18:
master. I cloned the repo on Sept. 16 and built from that. 

I can test from develop if you want me to.

rdmark, 2021-09-18:
@nsafran1217 If you have the time, it would be hugely helpful to us!

nsafran1217, 2021-09-18:
@rdmark Seems like its working correctly on develop as well. Formatted a new disk in Solaris 2.6 and did another fresh install of Solaris 2.6 on a new disk image.

I also tested formatting a new disk on Solaris 8 on a Sun Ultra 1 and it worked as well.

rdmark, 2021-09-18:
@nsafran1217 That's wonderful, very encouraging!  Let me close out this ticket.  If you find other issues, don't hesitate to raise a new ticket, or connect over at Discord.

closed 2021-09-18 by rdmark