PiSCSI/piscsi #87: 2.5" Powerbook SCSI drives

w0k3, 2021-03-14:
Hello,

are there any plans / did anyone try to support 2.5" SCSI drive emulation for the legendary Powerbooks / Duos? A Pi Zero could likely fit inside a Powerbook and Zero W could even enable wireless data (virtual drives or even actual network connection?).

akuker, 2021-04-13:
Definitely! The next version of RaSCSI (version 2.4) is being manufactured right now. Assuming it doesn't have issues, there should be kits available in early to mid May.

BleuLlama, 2021-04-13:
(that's super awesome!)

rdmark, 2021-08-20:
With 2.4 being available now, has any Powerbook owner tested that this combination works?

akuker, 2021-08-20:
You bet! I can confirm that it works in my PowerBook 520c.

The Raspberry Pi takes a while to boot, so the PowerBook will display the flashing question mark until the Pi boots. But, once the pi is booted, the Mac will find the drive and boot up.

There is a 3d model for a bracket here: https://github.com/akuker/RASCSI/tree/develop/hw/pb_bracket
(I'm not an artsy person, so its very utilitarian. But, no one will see it inside the case anyway!)

Also note, I can't speak for some of the newer PowerBooks, but my 520c's hard disk was thicker than what I've seen with a modern drive.  The bracket is around 17mm tall. So, you might want to check how much clearance you have in your laptop

closed 2021-09-15 by rdmark