PiSCSI/piscsi #100: Allow mounting drives on ID7

pacific202, 2021-05-07:
When mounting to a sampler the sampler is usually configurable for any ID (although the Mac connected to them in the 90's was usually ID7 it's rare to find a sampler that blocks this ID).

Could we please have the ability to mount a disk image on ID7?

uweseimet, 2021-08-09:
@pacific202 At least with the development branch I don't have issues using ID 7:

```
[2021-08-09 10:20:07.411] [info] +----+----+------+-------------------------------------
[2021-08-09 10:20:07.411] [info] | ID | UN | TYPE | DEVICE STATUS
[2021-08-09 10:20:07.411] [info] +----+----+------+-------------------------------------
[2021-08-09 10:20:07.411] [info] |  7 |  0 | SCHD | /home/us/hatari/test.hds
[2021-08-09 10:20:07.411] [info] +----+----+------+-----------------------------
```

Can you please verify whether your issue is still valid, and close it if it is not.

By the way, I also know platforms where drives can have ID 7. These are usually platforms that do not support SCSI initiator identification, i.e. where the initiator does not have an SCSI ID of its own.

pacific202, 2021-08-09:
I think this is something we need to add to the Web UI.

closed 2021-08-09 by pacific202