PiSCSI/piscsi #116: CHECK CONDITION not sent in case of SCSI command execution failures

uweseimet, 2021-06-23:
This issue was observed with a Pi 4, github revision 54cf432, RaSCSI board version 2.3. Computer (initiator) is an Atari TT with HDDRIVER 11 as hard disk driver and the SDRVTEST tool (https://github.com/uweseimet/atari_public/tree/main/SDRVTEST).

When rejecting unsupported SCSI commands (e.g. READ CAPACITY(16) with opcode 0x9e, but also others) the initiator does not receive CHECK CONDITION, even though SASIDEV::Status() is executed. There is also no CHECK CONDITION visible for the initiator when passing invalid block numbers (higher than the actual drive capacity) to READ(10). The logs do not show anything unusual.
There might be an problem with bus signal handling when sending the status byte.

Numerous real SCSI devices were tested in the same environment, and they all returned CHECK CONDITION as expected, which makes it likely that this is an issue with the RaSCSI software.

uweseimet, 2021-07-04:
Further analysis shows that CHECK CONDITION is actually sent, but the subsequent REQUEST SENSE returns a sense key of 0:

[2021-07-04 12:22:14.947] [trace] void SASIDEV::CmdRequestSense() REQUEST SENSE Command 
[2021-07-04 12:22:14.948] [trace] virtual int Disk::RequestSense(const DWORD*, BYTE*) size of data = 18
[2021-07-04 12:22:14.948] [trace] void SASIDEV::CmdRequestSense() Sense key $00

uweseimet, 2021-07-04:
This issue is a major one for software that evalutes the sense key and ASC in order to learn about details of a failed command. Unless I am missing something currently an error is reported (CHECK CONDIDTION), but when such a piece of software wants to react on it, REQUEST SENSE says that that was no previous error, which is contradictory.

@akuker I have prepared a fix for this issue, but without my pending PR being reviewed (and hopefully approved) I cannot really proceed, because the code base in my PR branch and in develop has started to differ too much.

akuker, 2021-07-05:
@uweseimet - I've closed the PR and pulled in the changes. Thank you for all of the great fixes!

uweseimet, 2021-07-05:
@akuker Thank you for merging my changes. Please see PR https://github.com/akuker/RASCSI/pull/120, which resolves this issue.

closed 2021-07-18 by uweseimet