PiSCSI/piscsi #126: Code is not 64 bit clean

uweseimet, 2021-07-08:
Compiler warnings indicate that there are issues with 64 bit environments. Ubuntu is available in a 64 bit version for the Raspberry Pi, and it would be nice if the software could be run on this OS.

uweseimet, 2021-07-17:
@akuker Regarding the gpiobus::bcm_host_get_peripheral_address problem there does not seem to be any newer code by Broadcom that supports 64 bit. But I assume that you know the address range occupied by your hardware in a 64 bit environment, don't you? Which address range is it?

uweseimet, 2021-08-04:
@akuker Would be great if you could comment on my question above.

akuker, 2021-08-05:
Hi @uweseimet - Unfortunately, I do not. I haven't really invested any time in researching what it would take to go to 64-bit. 

Are there benefits that we'd get by going to 64 bit? My thinking is that the GPIO reads/writes are probably the current bottleneck. I'm guessing going to 64-bit won't help that.

(Someday we will have to go to 64-bit, if/when the Raspberry Pi foundation drops 32 bit support)

akuker, 2021-08-05:
https://www.raspberrypi.org/forums/viewtopic.php?t=244031

uweseimet, 2021-08-05:
@akuker There is a benefit at least for those who would like to use their Pi not only for RaSCSI but additionally for other things, where a 64 bit OS may be useful. With 4 GB or more memory in your Pi 64 bit make sense. There is already a 64 bit Ubuntu for the Pi, by the way.

closed 2021-08-21 by uweseimet