PiSCSI/piscsi #131: No usable documentation on initiator mode

cm68, 2021-07-13:
the ability to inquire/read/write/format scsi devices connected to the PI via scsi commands would be extremely useful.
the variety of boards supported by this software has varying levels of initiator support.

this really needs to be documented.

akuker, 2021-07-17:
Agreed. I have only tried to use the initiator functionality once, and it didn't work. rasdump is what *should* be able to dump the contents of a drive.

uweseimet, 2022-08-26:
Isn't this related to https://github.com/akuker/RASCSI/issues/454 and thus can be closed because initiator mode cannot be supported?

akuker, 2022-08-26:
initiator mode CAN be supported. Rasdump is different than the functionality described in #454. Rascump simply dumps the drive contents to a disk image. Where #454 is wanting it to be a full fledged SCSI interface.

uweseimet, 2022-08-27:
@akuker In https://github.com/akuker/RASCSI/issues/454 you say that initiator mode support can only be implemented with a huge effort, and only with a single device. On the other hand the Wiki says that the FULLSPEC board supports initiator mode, and in this case documentation on it would be important, because the RaSCSI projects advertises this feature.
I (and maybe others) may be misunderstanding something. Can you please elaborate in order to resolve my confusion?

uweseimet, 2022-11-06:
@cm78 As part of fixing code issues with rasdump (lots of SonarQube warnings, and half of the rasdump code is redundant/duplicated) I analyzed the rasdump code and I think that it provides all the information you need to know. Essentially, the bus has to be initialized with BUS::mode_e::INITIATOR and then you can send commands just like rasdump does, by setting/checking ths bus signals.
A quick test with a streamer reveals that sending an INQUIRY command works at least for some devices:
```
>sudo ./bin/fullspec/rasdump -i 3 -b 7 -f dumpfile
RaSCSI hard disk dump utility version 22.10.1 (Nov  6 2022, 11:01:41)
TARGET ID               : 3
BOARD ID                : 7
Vendor                  : HP      
Product                 : HP35470A        
Revison                 : 1109
READ CAPACITY ERROR -3
```
For one of my hard disk drives rasdump fails, but I don't know why. But anyway, as far as documentation is concerned, SCSI bus signals are described in the SCSI specification, i.e. they are not rascsi-specific. Saying that I would like to close this ticket, because the rasdump code and the SCSI specification should provide all required information.

uweseimet, 2022-11-09:
@cm68 My rasdump fixed and improvements are in the develop branch now, so that they will become part of the next (after 22.10.01) release. rasdump is fully working now and shows how to use RaSCSI's initiatiator mode.
Thus closing this ticket.

closed 2022-11-09 by uweseimet