PiSCSI/piscsi #136: Improve command line argument parser

uweseimet, 2021-07-19:
IMO the current parser has potential for improvements, in particular the handling of the ID option. In case of a better LUN support (SCSI supports at least 8 LUNs) this option would require one more argument, i.e. not just the filename.
My suggestion is to change the syntax of this option to require exactly one (compound) argument like this:

-ID ID,FILENAME

and with maybe better LUN support later to:

-ID ID:LUN,FILENAME

As far as I can tell this simplifies the parser and makes future extensions easier.

@akuker I would like to hear your opinion on this.

akuker, 2021-07-19:
@uweseimet - I agree that the current parser has a lot of room for improvement.

In general: I have a personal goal of keeping the external interfaces compatible with the original RaSCSI application as much as possible. (Adding new things is fine thought)

For better LUN support, appending that to the SCSI ID sounds great and shouldn't break compatibility
`- ID<ID>:<LUN>`

Similarly you could do the same thing with the image type:
`rascsi -ID1:0 /home/pi/images/install.bin:cd`

Note that the parent RaSCSI project added the ability to save/load files with the configuration in version 1.50. That's something that I'd also like to bring over at some point.

I've also wanted to somehow make the rasctl and rascsi command line argument syntax line up a little more. Not sure how to do that without breaking backwards compatibility though.

I've also wanted to add the option to override some of the default configuration options, such as vendor name.
`rascsi -ID1:0 /home/pi/images/install.bin --options type=scsihd,vendor=IBM,product=DESKSTAR,revision=1.22`

.... In summary, I'm rambling... I'm open to changes. I have a goal of maintaining compatibility between the two different forks, but that's a _goal_. 

I'm very curious to hear what your thoughts are on this!

uweseimet, 2021-07-19:
@akuker Keeping the compatibility should work by just keeping the existing options and adding new ones that essentially provide a different mapping, just as you indicated.
Regarding configuration files it would be good if any command line settings could override the configuration file settings and could add settings not present in the configuration files.
I also agree with configurable vendor names, which is something I was also missing.

uweseimet, 2021-07-19:
I'm just wondering: A new parser would probably support multiword-arguments and shortcuts like, e.g. --enable-my_option/-m, which differs quite a lot from what the current parser expects. One way to resolve this and be backwards compatible might be to just say: If the command line contains the "-ID" or "-HD" argument the old parser is used (backwards compatible), and if there is no such argument the new parser is used.

akuker, 2021-07-22:
I like the backwards compatibility mode idea. But, that just adds technical debt.

What do you think about having the backwards compatibility mode for a few months, then yanking it out?

uweseimet, 2021-07-22:
Yes, this should be considered.

I played around a bit with how a modified parser might look like. The main issue I stumbled upon is the fact that the existing parser accepts the same options several times. This is of course in order to be able to define several devices on a single command line, but from what I can tell this forces the parser code to always be rather awkward. On the other hand, being able to define more than one ID is convenient, because you are not forced to use a configuration file.
Sometimes the command line is convenient, sometimes a configuration file is ;-).

uweseimet, 2021-08-30:
I am going to close this issue because options for providing vendor, product, revision and type have been implemented. If we need an option update for dealing with SCSI LUNs other code has to be changed anyway, and then the parser can be updated if needed.

closed 2021-08-30 by uweseimet