PiSCSI/piscsi #137: Device list should not be a string but a data structure

uweseimet, 2021-07-19:
Currently the device list is returned as a (pre-formatted) string, which is fine for console output. Potential other (non-console) clients would run into issues parsing this output. The output should not be formatted on the server but on the client side, depending on the nees of the client.

I'd like to suggest to extend the protobuf interface by a data structure used for returning the device list. There is already a special handling for the device list right now; the list is special anyway compared with other commands, which just return a status and a typically one-line string.

@akuker As always, your thoughts on this are welcome.

akuker, 2021-07-19:
I agree with this idea. The output of rasctl will need to be _identical_ though. The web interface and OLED script both rely on that formatting. It shouldn't be too hard to just copy the formatting logic from rascsi.cpp to rasctl.cpp (or move it to a shared file). 

I think we should keep the table being printed to the log file, but the formatting of that isn't as restrictive. I'm not aware of anyone that's parsing the log file currently.

Protobuf would be fantastic for passing around the data structure.

uweseimet, 2021-07-19:
I agree that the output of rasctl has to stay as it is. At some time in the future the web interface might directly talk to the rascsi process, e.g. with JavaScript, for which a protobuf code generator is available: https://developers.google.com/protocol-buffers/docs/reference/javascript-generated. This would decouple the web interface from the command line client.

I am going to start with the required changes for this issue.

akuker, 2021-07-19:
Awesome! Thank you for all of the awesome changes that you've proposed!!

closed 2021-07-24 by uweseimet