PiSCSI/piscsi #138: Support a pool of available drive images

uweseimet, 2021-07-19:
Typcially most images reside in the same folder. It would be convenient for the users if one could specify those images just by their name instead of their full path. In addition, the set of available images might be displayed by the web interface so that one could simply select new images from this image pool.
Essentially, a new command line option that specified the path of this image folder is needed, with some simple logic to look up the images in the pool.

akuker, 2021-07-19:
The web interface currently does show a listing of the available images in the /home/pi/images directory. We could use that as the default, then add a command line option to override that at startup.

uweseimet, 2021-07-19:
Sounds good!

uweseimet, 2021-08-04:
The location of the default image folder should be known to the rascsi process, so that one can use rasctl to attach an image without providing the full path.
The default image folder should be returned with PbServerInfo.

closed 2021-08-06 by uweseimet