PiSCSI/piscsi #139: Discard current baremetal support

uweseimet, 2021-07-19:
@akuker Correct me if I am wrong: You indicated that you are not interested in baremetal support as it is right now, but favor a different solution. In this case the existing code would profit from all those baremetal ifdefs to be removed. This would definitely improve code maintainability.

akuker, 2021-07-19:
Agreed. Let's just make that executive decision now.... Baremetal support can be removed.

If someone really needs baremetal support, they can go with the GIMONS version of the code.

*Maybe* someday we add it back in... but even then I would probably use a FreeRTOS or something instead of just bare metal.

uweseimet, 2021-07-19:
Cool! I will address this issue next, because any other changes will profit from the code having been cleaned up first.

closed 2021-07-20 by uweseimet

uweseimet, 2021-07-20:
Resolved with https://github.com/akuker/RASCSI/commit/3c5a0003a43fd3efd16c14e23918e8865a4f77cc