PiSCSI/piscsi #144: Web interface should use protobuf interface and not depend on rasctl

uweseimet, 2021-07-22:
Instead of relying on rascsl (output format and command syntax) the web interface should directly use the protobuf interface. A code generator for protobuf is available for Python, see https://developers.google.com/protocol-buffers/docs/pythontutorial.

uweseimet, 2021-09-01:
@rdmark I'd suggest to generate it on the fly, similar to how it is done for the generated C++ code.

rdmark, 2021-09-02:
@uweseimet I made some progress, generating the Python pb2 library, creating proto objects in Python code, and serialize them.  The one thing that isn't clear to me is how to pass the serialized message to the rascsi protobuf interface?  The google dev tutorial and others mostly demonstrate how to write the serialized message to a file, others how to use TCP/IP to transmit it, but I'm not sure either is applicable here. 

BTW this is how to compile the Python pb2 library:
`protoc -I=/home/pi/RASCSI/src/raspberrypi/ --python_out=/home/pi/RASCSI/src/web/ rascsi_interface.proto`

And this goes into requirements.txt:
`protobuf>=3.17.3`

rdmark, 2021-09-02:
For instance I believe this would be a working implementation of detach_all() using protobufs:
```
import rascsi_interface_pb2 as proto
def detach_all():
    command = proto.PbCommand()
    command.operation = proto.PbOperation.DETACH_ALL
    command.SerializeToString()
```

The question is how to get the serialized message to rascsi for deserialization. :)

uweseimet, 2021-09-02:
@rdmark You have to write the serialized string to a socket and receive the result from the same socket. First the header (see the comment in rascsi_protobuf.proto), then the actual data.
See SendCommand() in rasctl.cpp and the code in protobuf_util.cpp for the equvialent code in C++.

rdmark, 2021-09-03:
I got it to work at a basic level!  WIP feature branch here https://github.com/rdmark/RASCSI/tree/feature_webapp_protobuf

rdmark, 2021-09-03:
All commands have been reimplemented using protobuf now, and functionality seems unchanged from before.
I've also added some error handling for sockets and protobuf response.

Currently trying to figure out how to capture the response correctly. When deserializing the response I don't seem to get the object that I expect, e.g. PbResult.

rdmark, 2021-09-04:
I was able to capture and parse the responses properly now, and implemented the get_type() method with protobufs.
What's left to do is implement the list_device() method, and we should be entirely independent of rasctl for the web ui!

Also need to improve error handling, show correct result messages in the UI, and do a ton of cleanup.

rdmark, 2021-09-05:
@uweseimet @akuker @erichelgeson Please review this PR draft. https://github.com/akuker/RASCSI/pull/229
The behavior of the webapp should be generally unchanged, with some minor differences in device names displayed.

closed 2021-09-15 by uweseimet