PiSCSI/piscsi #148: Replace DWORD disk ID by a string

uweseimet, 2021-07-23:
I cannot find any reason why the DWORD type is used to store device IDs in disk.cpp/disk.h. Using a string would be more straightforward.
Ultimately an object-oriented approach would provide even more improvements.

@akuker In the feature_id_improvements branch I formally replaced the DWORD by a string. I would like to hear your opinion on changing the ID data type.

akuker, 2021-07-24:
As long as it doesn't have a huge performance it (I don't think it will), I'm good with that change. I'm wondering if an enum would be more appropriate? But string is fine too. I approved the PR.

Thank you for another awesome pull request, @uweseimet

uweseimet, 2021-07-24:
Yes, an enum would be better. Actually, the PbDeviceType enum would be the best candidate. But due to the old protobuf version on Raspbian, which does not support getting the enum value as a string, we cannot simply use it. This is why at least for now I just a string, similar to how it was before.

@akuker Thank you for merging everything.

closed 2021-07-24 by uweseimet