PiSCSI/piscsi #153: SCSI command 4A is not implemented - Get Event Status Notification

akuker, 2021-07-27:
Using the current develop branch, when a modern Ubuntu host is used, the RaSCSI logs are filled with the following error messages:

```
Jul 26 19:35:51 rascsi RASCSI[30336]: [2021-07-26 19:35:51.768] [warning] virtual void SCSIDEV::Execute() Received unsupported command: $4A
Jul 26 19:35:51 rascsi RASCSI[30336]: [2021-07-26 19:35:51.768] [warning] void SASIDEV::CmdInvalid() Command not supported
```

The CD-ROM functionality still seems to be functional. On older Macintosh hardware, this error message isn't seen. But with Ubuntu 20, it is logged approximately every 2 seconds.

Command 4A is "GET EVENT STATUS NOTIFICATION". Details of how this should be implemented are in section 9.1.2 of the following document:
https://www.t10.org/ftp/t10/document.97/97-108r0.pdf

uweseimet, 2021-08-30:
Note that the latest document from 2009 (like the other SCSI specification documents, which are still being worked on) are available on https://www.t10.org/drafts.htm.
I already added a naive implementation some time ago, so that the logs are not flooded with warnings. If notification by polling is requested a message on trace level will be logged with the latest codebase in uweseimet_develop. Requests for asynchronous notification are rejected because this kind of notification is optional.

uweseimet, 2023-10-16:
Is there still any need for this ticket, i.e. for more than the naive implementation I added? If not I suggest to close this ticket. There are numerous SCSI commands PiSCSI does not implement, and IMO we should only add a ticket for a missing command if without this command something important does not work.