PiSCSI/piscsi #185: Modular device implementation approach (dynamic loading of devices)

uweseimet, 2021-08-10:
Probably not easy to implement (definitely requires better object-oriented code than there is now), but I like the idea.
When a new device is created, based on its type the respective implementation module should be loaded on demand.

uweseimet, 2022-02-12:
@akuker This ticket is mainly meant to be an incenctive for improving modularity. But I'm find with closing it.

akuker, 2022-02-12:
I have no objections with improving modularity. Also no objections with dynamic libraries if it helps solve a problem :)

uweseimet, 2022-02-12:
@akuker In this case let's keep it open for now. The idea was, that if you can implement dynamic loading of devices you have reached a decent level of modulariy, as far as devices types are concerned. We are not yet there, but https://github.com/akuker/RASCSI/pull/662 is a step forward.

uweseimet, 2022-08-27:
Closed as part of a ticket cleanup. There are too many tickets already, a lot of them half-started or abandoned or in some other kind of undefined state.

closed 2022-08-27 by uweseimet