PiSCSI/piscsi #191: DaynaPort not working in branch develop

pfuentes69, 2021-08-15:
Current develop branch fails for DaynaPort and auto mount. While accessing the DP from the Mac, RASCSI log shows a warning "Received a CmdRetrieveStats command for a non-daynaport unit SCDP". Probably all related to the new device management. 

Branch daynaport3 still works fine, not having these 2 issues. 

If I may do a suggestion, it would be good to have a dedicated branch named "experimental" that is stable enough (at least supporting DP and auto mount) and not subject to ongoing works.

uweseimet, 2021-08-15:
@pfuentes69 Most likely fixed in the uweseimet_develop branch. Would be great if you could test whether the problem is gone with this branch.
In general, it's the nature of a develop branch that it is not necessarily stable and free of issues.

uweseimet, 2021-08-15:
@pfuentes69 By the way, do you have any instructions how to use the DP on a 68K Mac (Performa 630)? I'm not a Mac user, and only use this Mac for testing, but I don't know how to install the DP with it.

pfuentes69, 2021-08-15:
@uweseimet Hi!
I won't be able to test today, but certainly I'll do it ASAP. 

The main issue is that the instructions to setup the DaynaPort (https://github.com/akuker/RASCSI/wiki/Dayna-Port-SCSI-Link) tell people to use the develop build, which logically is not meant to be stable, so this is tricky. 

That's why I suggested to create an "experimental" branch, which is kind of an "stable develop"... at least until DaynaPort is part of the master branch.

pfuentes69, 2021-08-15:
> @pfuentes69 By the way, do you have any instructions how to use the DP on a 68K Mac (Performa 630)? I'm not a Mac user, and only use this Mac for testing, but I don't know how to install the DP with it.

Please check the link in the above comment. These are the best instructions so far, but right now it will only work with the daynaport3 branch, not the develop branch as stated in the wiki

pfuentes69, 2021-08-15:
BTW... should the auto mount work again too with the latest develop?

uweseimet, 2021-08-15:
@pfuentes69 Now I remember actually reading these instructions, but it's inconvenient for me to use my Pi based on these instructions, because for my Mac I use a Pi that is usually not connected to the LAN but only to the WLAN.

pfuentes69, 2021-08-15:
> @pfuentes69 Now I remember actually reading these instructions, but it's inconvenient for me to use my Pi based on these instructions, because for my Mac I use a Pi that is usually not connected by ethernet by only by Wifi.

A guy posted instructions in Discord yesterday to setup on wireless. 

https://docs.google.com/document/d/1I4iFgQSYBCAi0xo9GqFOMDDFmt683b2TNaBw9j1fkb4/edit#

I didn't test myself though...

uweseimet, 2021-08-15:
I'm afraid I don't know. I have no idea when the daynaport branch was merged into develop the last time, and which changes have only been applied to the daynaport branch in the meantime.
The problem with the daynaport code is that this code inherits from the disk code, even though it is not a disk, and that there are hard-coded daynaport checks in other classes. This is prone to side-effects.

rdmark, 2021-08-15:
Thanks for raising this issue in the tracker, pfuentes69.  (I'm the guy who wrote the wifi setup guide :) )
I'm compiling the latest revision of uweseimet_develop (ec9216a) on my Pi now, and will report back how well it works with my Macs.

uweseimet, 2021-08-15:
@rdmark Cool! Note that if there are is an issue it might have been there even earlier than my commits to the develop branch. Anyway, by comparing the DP code (it's not that much) with the daynaport3 branch one should be able to identify any relevant divergency.

rdmark, 2021-08-15:
Agreed, I was looking at a diff between develop and daynaport3 this morning, and the changeset didn't look too overwhelming.  Should be possible to figure it out relatively quickly, unless there is a ton of coupling elsewhere in the codebase.

uweseimet, 2021-08-15:
The fact that different device types (hard disk, bridge, daynaport) are not cleanly separated is one of the issues I have been working on. But a clean separation still requires a lot of changes, and I don't think it will be possible very soon.

pfuentes69, 2021-08-15:
> Thanks for raising this issue in the tracker, pfuentes69.  (I'm the guy who wrote the wifi setup guide :) )
> 
> I'm compiling the latest revision of uweseimet_develop (ec9216a) on my Pi now, and will report back how well it works with my Macs.

Hi "guy" ;)

I'll be out from home for several days, but I'll try to test too when back. I'll check for your results!

For me the basic need is auto mount and DaynaPort setup... let's see if Uwe could put that on tracks again. 

Pedro

rdmark, 2021-08-15:
@uweseimet So I tested on two of my vintage Macs, one using the legacy MacTCP 2.1 networking stack, one using the more modern Open Transport 1.3 stack.  Unfortunately, when trying to ping any IP address, it fails with this message in the logs:

> Aug 15 23:20:16 raspberrypi RASCSI[12722]: [2021-08-15 23:20:16.359] [info] The DaynaPort interface has been ENABLED.
> Aug 15 23:20:16 raspberrypi kernel: rascsi_bridge: port 1(ras0) entered blocking state
> Aug 15 23:20:16 raspberrypi kernel: rascsi_bridge: port 1(ras0) entered forwarding state
> Aug 15 23:20:17 raspberrypi RASCSI[12722]: [2021-08-15 23:20:17.087] [warning] Received 'Mode Select'
> Aug 15 23:20:17 raspberrypi RASCSI[12722]: [2021-08-15 23:20:17.087] [warning]    Operation Code: [0A]
> Aug 15 23:20:17 raspberrypi RASCSI[12722]: [2021-08-15 23:20:17.087] [warning]    Logical Unit 0, PF 0, SP 0 [00]
> Aug 15 23:20:17 raspberrypi RASCSI[12722]: [2021-08-15 23:20:17.087] [warning]    Reserved: 00
> Aug 15 23:20:17 raspberrypi RASCSI[12722]: [2021-08-15 23:20:17.088] [warning]    Parameter List Len 2A
> Aug 15 23:20:17 raspberrypi RASCSI[12722]: [2021-08-15 23:20:17.088] [warning]    Reserved: 80
> Aug 15 23:20:17 raspberrypi RASCSI[12722]: [2021-08-15 23:20:17.088] [warning]    Ctrl Len: 00000000
> Aug 15 23:20:17 raspberrypi RASCSI[12722]: [2021-08-15 23:20:17.088] [warning] Error occured while processing Mode Select command 0A

On a positive note, the interface is able to get initialized and it can resolve its own IP address, and the Daynaport hardware test passes.  So it seems to work 'better' than on the latest develop branch code.  But as soon as you try to ping any IP (such as the bridge on the Pi, or a public Google DNS server) you get timeouts or error 500s, while the above "Error occured while processing Mode Select command 0A" warning block appears over and over in the logs.

rdmark, 2021-08-15:
As a side note, my 600MB hard drive image that I created as part of the initial setup script (default settings) doesn't work well on this latest code in uweseimet_devel, when mounted as a non-system disk. Mac OS (7.6.1) keeps complaining that the file system is damaged over and over until I dismount it. The very same image worked fine with develop and daynaport3 code. Perhaps something's you're already on top of, but I thought I'd mention it.

uweseimet, 2021-08-16:
@rdmark Thank you for testing. I also ran more tests today and fixed some (not only new) issues, e.g. that "Received Mode Select", was reported on write/verify commands and not only on mode select itself (see the
uweseimet_develop diffs if you are interested).
There was also an issue with the new write protection feature for regular hard disk media, which may have been the cause for your Mac reporting a damaged file system.

With the latest uweseimet_develop codebase my Mac is fine. I tested with a 600 MB hard disk, copied applications to it and could successfully start them.
If you still have issues, a logfile excerpt on "trace" log level would be great. Just start rascsi with the "-g trace" option to set the log level, or use "rasct -g trace" to change the log level for a running instance of rascsi just before you start the operation which should be logged in detail.

Regarding the daynaport, I fixed a potential issue but (at least with my current setup) cannot test it. If possible, please give it a try. At least the bogus "Received Mode Select" should be gone.

uweseimet, 2021-08-16:
@rdmark In the meantime I installed OpenTransport and the DaynaPort drivers and set up the network as explained on your page. rascsi display a lot of DaynaPort-related trace output, but I'm afraid I don't know how to finalize my setup. I'm not familiar with Mac networking, AppleTalk etc. I'm afraid.

rdmark, 2021-08-16:
@uweseimet I've tried building 21c726e and can confirm that the "Received Mode Select" messages have been suppressed. I'm still unable to ping any IP from the Mac, including the bridge interface itself. Each request times out almost instantly.

Checked out & building 0f0c657 now (which takes forever on a Pi Zero -- I think it overheats very quickly and slow down when compiling code) to see if that makes a difference. If not, I'll play around with the log levels and see if I can get some insights where the requests get hung up.

Thank for trying to set up a Mac environment to test yourself. Did you choose the right Ethernet interface in the TCP/IP control panel? The emulated DaynaPORT interface is usually called "Alternative Ethernet". If you have wired Ethernet on your Pi, you can also try the ['official' method](https://github.com/akuker/RASCSI/wiki/Dayna-Port-SCSI-Link).

uweseimet, 2021-08-16:
I just succeeded in getting my Mac network with DP to work, after switching from DHCP to manual configuration. I somehow figured out, just as you said, that I need the alternative ethernet. Using the Netscacpe Explorer I can now access hosts in my local network. External addresses also work, but whenever I enter a nameserver address in the configuration dialog the system hangs up immediately.
But except for ths nameserver configuration issue I don't have any obvious issues.

Yes, building on a Pi Zero is slow. I usually use the Eclipse IDE on a regular Linux x86 PC for time-consuming changes (this also gives you refactoring support etc.) and then check out on the Pi (usually a Pi 4) for compling and testing on the real hardware. And I usually run rasctl on my x86 PC and connect remotely to the Pi with "rasctl -h HOSTNAM". All in all one can do a lot of the development work in a powerful environment. 

https://github.com/akuker/RASCSI/commit/0f0c6573334064f69ab37c27a1b824c4b8b8722a should be fine, at least it is for me.

Any idea what might cause my nameserver issue?

rdmark, 2021-08-16:
I'm back online again, so I think 0f0c657 is a winner (as far as DP is concerned)!

[image]

uweseimet, 2021-08-16:
:-) Great news! Can you please tell me how your TCP/IP configuration looks like on the Mac? I would like to try to resolve my nameserver problem.

rdmark, 2021-08-16:
@uweseimet This is what my MacTCP configuration looks like. Give me a few min to switch over to my OT setup. I should make sure this code works over there too.

rdmark, 2021-08-16:
[image]

uweseimet, 2021-08-16:
@rdmark Looks as if you are using a different tool for setting up the nameserver IP. I will try that one later most likely.

pfuentes69, 2021-08-16:
> @rdmark Looks as if you are using a different tool for setting up the nameserver IP. I will try that one later most likely.

It seems you're using OpenTransport. 
To see that up config screen you'd have to use the network selection tool in "Apple extras" to switch the networking to "Classic Networking"

uweseimet, 2021-08-16:
@pfuentes69 Can you also confirm that with the current uweseimet_develop branch your DP is working again?

rdmark, 2021-08-16:
@uweseimet You should only really use MacTCP / Classic Networking on really old machines that can't use OT (68000 Macs, System 7.1 or earlier).  Here's what my OT setup looks like.  It works great too!
[image]

pfuentes69, 2021-08-16:
> @pfuentes69 Can you also confirm that with the current uweseimet_develop branch your DP is working again?

Sorry but I'm in a business trip for part of the week. I could try Thursday probably.

rdmark, 2021-08-16:
@uweseimet BTW, the 'damaged file system' errors seem to have been resolved as well. I can copy files back and forth just fine (although just cursory testing so far.)

uweseimet, 2021-08-16:
@rdmark Sounds good! Regarding the network setup my system is 7.6.on a Performa630. I don't have installed MacTCP you, but will most likely try later. For now (i.e. for testing) local connections without nameservers are sufficient for me.

uweseimet, 2021-08-16:
@pfuentes69 OK. I hope that till then uweseimet_develop has already been merged into the develop branch.

rdmark, 2021-08-16:
@uweseimet I booted into my 7.6.1 drive w/ OT 1.3 and could confirm that it works perfectly there too, including DNS lookup.  Have you tried using the helpful little network troubleshooting tool called OTTool? If not, grab a [copy from the Garden](https://macintoshgarden.org/apps/ottool-121).

uweseimet, 2021-08-16:
@rdmark Your nameserver setup also works for me. Before, I just added a single IP with no CR/LF at the end. Now I added two and (just in case) a CR/LF after the second, and it worked. So either the software requires the CR/Lf or it requires two IPs.

rdmark, 2021-08-16:
@uweseimet That's an interesting quirk! I attached these screenshots to my doc to help others avoid that pitfall.

uweseimet, 2021-08-16:
@rdmark I checked this and found that the system hangs up when you enter just one IP instead of two. CR/LF does not matter.

rdmark, 2021-08-16:
@uweseimet The one issue that I see remaining right with WRT daynaport, is that the old syntax for attaching devices when the service starts up doesn't seem to work with the daynaport device in particular.

If you say put this in your rascsi.service [Service] section, you get an error when the service starts up.
`ExecStart=/usr/local/bin/rascsi -ID4 daynaport `

> Aug 16 21:12:40 raspberrypi RASCSI[2899]: [2021-08-16 21:12:40.615] [info] Validating: cmd=ATTACH, id=4, unit=0, type=UNDEFINED, filename='daynaport', device name='', params=''
> Aug 16 21:12:40 raspberrypi RASCSI[2899]: [2021-08-16 21:12:40.618] [info] Executing: cmd=ATTACH, id=4, unit=0, type=UNDEFINED, filename='daynaport', device name='', params=''
> Aug 16 21:12:40 raspberrypi RASCSI[2899]: [2021-08-16 21:12:40.620] [warning] Received a command for an invalid device type: UNDEFINED
> Aug 16 21:12:40 raspberrypi RASCSI[2899]: Received a command for an invalid device type: UNDEFINED

Did the syntax here change in current development code? Note that the old syntax for doing this for a hard disk image still works fine.

uweseimet, 2021-08-16:
@rdmark Looks as if I missed that when testing my changes for backwards compatibility. I will address this asap.

uweseimet, 2021-08-16:
@rdmark I just commited a fix to uweseimet_develop. I could not test it with real hardware, but verified that now rascsi reports that a daynaport device was attached when using "daynaport" as a filename, which it did not do before this change.

rdmark, 2021-08-17:
@uweseimet Testing cdb2890, I can get the daynaport device to attach on startup. Solid fix!

BTW when you say backwards compatibility, does this mean that there's a new syntax that we should be using in the service configuration moving forward?

uweseimet, 2021-08-17:
@rdmark I added some extensions to rascsi and rasctl, in order to make everything more flexible. It's probably best if your read the current manpages (in uweseimet_devel) for an overview. The old syntax is still supposed to work, though.

rdmark, 2021-08-17:
@uweseimet Nice, I love man pages. I had a quick look, and I assume you mean the syntax that looks something like this?
`ExecStart=/usr/local/bin/rascsi -i 3 -c attach -t schd -f /home/pi/images/600MB.hda`

I wasn't actually able to get this syntax to work from within the rascsi.service configuration, although it works great with rasctl on the command line.  It throws "/usr/local/bin/rascsi: invalid option -- 'c'" and then exits.  Am I using it wrong?

uweseimet, 2021-08-18:
@rdmark Note that the syntax of rascsi and rasctl differs (this difference is not related to my changes, it was always there). For each command the manpage explains the respective syntax. For rascsi it is something like

```
rascsi -ID 3 -t schd /home/pi/images/600MB.hda
```

pfuentes69, 2021-08-18:
Hello. Just to fully understand... what would be the syntax to configure the service to auto mount both a disk and a DaynaPort?

uweseimet, 2021-08-18:
@pfuentes69 Since the syntax is the same as it was before my changes it is

```
rascsi -ID 6 dynaport
```

rascsi then displays:

```
[2021-08-18 09:08:44.577] [info] Added new SCDP device, ID: 6 unit: 0
+----+----+------+-------------------------------------
| ID | UN | TYPE | DEVICE STATUS
+----+----+------+-------------------------------------
|  6 |  0 | SCDP | DaynaPort SCSI/Link
+----+----+------+-------------------------------------
[2021-08-18 09:08:44.577] [info] +----+----+------+-------------------------------------
[2021-08-18 09:08:44.577] [info] | ID | UN | TYPE | DEVICE STATUS
[2021-08-18 09:08:44.577] [info] +----+----+------+-------------------------------------
[2021-08-18 09:08:44.577] [info] |  6 |  0 | SCDP | DaynaPort SCSI/Link
[2021-08-18 09:08:44.577] [info] +----+----+------+-------------------------------------
```

uweseimet, 2021-08-18:
Just for completeness sake: With the new -f option you can tell rascsi the default folder for images, which shortens the command line if you have several images, e.g.

```
rascsi -f /home/us/images -ID0 scsi.hds -ID1 test.hds -ID2 test.mos -ID5 test.iso
```

```
+----+----+------+-------------------------------------
| ID | UN | TYPE | DEVICE STATUS
+----+----+------+-------------------------------------
|  0 |  0 | SCHD | /home/us/images/scsi.hds
|  1 |  0 | SCRM | /home/us/images/test.hds
|  2 |  0 | SCMO | /home/us/images/test.mos
|  5 |  0 | SCCD | /home/us/images/test.iso (WRITEPROTECT)
+----+----+------+-------------------------------------
```

rdmark, 2021-08-19:
@uweseimet This is a bit off topic, but I couldn't think of a better way to quickly make you aware (without raising a new issue ticket):  In the latest uweseimet_develop 8091963 I'm no longer able to attach either of my two HD images.  The Mac tried to initialize them, while the log outputs dozens of these messages:
> [warning] ID SCHD: Media capacity of 9 blocks exceeded: Trying to read block 96, block count 2
or
> [warning] ID SCHD: Media capacity of 9 blocks exceeded: Trying to read block 98, block count 1

pfuentes69, 2021-08-19:
Hi guys,
I'm back home and I was about to try this, but as per the last comment I understand something is broken. 
I'll stay tuned then...

uweseimet, 2021-08-19:
@rdmark I'm afraid I cannot reproduce this. My Mac is fine with a 600 MB drive created by the web interface. Can you please run "make clean" and then rebuild with "make" and try again?

uweseimet, 2021-08-19:
@rdmark A different question, related to the DP again. I created a branch "feature_wlan_bridge", where when eth0 is not up these commands from your instructions are executed programmatically:

```
sudo brctl addbr rascsi_bridge
sudo ifconfig rascsi_bridge 10.10.20.1/24 up
```
This way rascsi essentially deals with the wlan0 interface as it deals with the eth0 interface.

Do you think this is useful, and does it make sense to do this when eth0 is not up, or would there be a better criteria?

pfuentes69, 2021-08-19:
Hello. I confirm the "Media capacity" errors... The images mount, but I got a sad Mac.
Something is broken...

uweseimet, 2021-08-19:
@pfuentes69 Hi. Since I cannot reproduce any errors with my Mac, can you please tell me the exact size (in bytes) of the non-working image file(s)?

rdmark, 2021-08-19:
@uweseimet I tried building acbeed1 from your dev branch but still the same problem. Both of my hda images are 629145600 bytes, the default 600MB image that the easyinstall.sh script creates.

pfuentes69, 2021-08-19:
@uweseimet 
I typically mount two images, that work fine in other branches of RASCSI and also in BlueSCSI.
My Macs gives these sizes:
System: 524.288.000 bytes (524,3 MB on disk)
Software: 2.147.483.648 bytes (2,15 GB on disk)
The Mac couldn't boot, so I guess the big image is irrelevant. This system image, compressed is 40MB so I just sent it to you with a WeTransfer link.
Thanks!

uweseimet, 2021-08-19:
@pfuentes69 Sounds good, I will test with your image.

uweseimet, 2021-08-19:
@pfuentes69 Strange, my Mac has no issues with accessng your image. I did not boot from it (don't know whether a Performa630 can boot from SCSI and if it can, how), just mounted it and my Mac can display its contents.

pfuentes69, 2021-08-19:
Well... I can only say that these images work in the branches master, daynaport3 and the develop as it was last weekend.
@rdmark also had the issue just now... I don't know if he can tell you exactly the last version that was working...

uweseimet, 2021-08-19:
@pfuentes69 Yes, I don't doubt that something is wrong, but I cannot reproduce it ... Would be great if somebody knew which commit caused this issue.

pfuentes69, 2021-08-19:
I just left home... I'll have access to the Mac later, sorry. 
I saw in the logs a very similar error than the one @rdmark said... similar to "[warning] ID SCHD: Media capacity of 9 blocks exceeded: Trying to read block 98, block count 1", but maybe with different values.

BTW In theory (no idea if this works with the P630) to boot from a particular SCSI ID you'd keep pressed at boot: Command-Option-Shift-Delete-# (where # is the number of the SCSI ID)

uweseimet, 2021-08-19:
@pfuentes69  Thank you, I'm currently reading https://poynton.ca/notes/mac/Mac_startup_keys_obsolete.html, to learn about all the available start up keys.

rdmark, 2021-08-19:
For the record, I tested on both my Performa 476 and SE now, both as startup disk and as regular disk. The exact same issue on both systems.

[33 more replies]

closed 2021-08-22 by pfuentes69