PiSCSI/piscsi #195: Allow attaching a CD ROM drive without an image

rdmark, 2021-08-20:
# Info

- Which version of Pi are you using: Pi Zero W
- Which github revision of software: b32f324
- Which board version: 2.4a
- Which computer is the RaSCSI connected to: Mac SE

# Describe the issue

It would be convenient to be able to attach a CD ROM drive without specifying an image to mount with it.
The benefit of having an 'empty' CD ROM drive is that inserting an image during runtime would allow you to immediately mount it in the OS, without having to go out of your way to use a tool or reboot the machine to recognize the new SCSI device.

Currently you get something like:
> rasctl -c attach -t sccd -i 3                                                                        
> Error: Device type SCCD requires a filename

rdmark, 2021-08-20:
Something like this in rascsi.service doesn't work either. The option just gets ignored.

> ExecStart=/usr/local/bin/rascsi -ID5 -t sccd

rdmark, 2021-08-20:
Additionally, by allowing you to attach a device by simply its type, you may also get these shorthands for attaching Bridge or Daynaport devices (neither of which current work):
> rascsi -ID4 -t scbr
> rascsi -ID4 -t scdp

uweseimet, 2021-08-20:
@rdmark A change in the latest uweseimet_develop branch should resolve the problem for rasctl:

```
>rasctl -c a -t sccd -i 3
>rasctl -l
+----+----+------+-------------------------------------
| ID | UN | TYPE | DEVICE STATUS
+----+----+------+-------------------------------------
|  3 |  0 | SCCD | NO MEDIA (WRITEPROTECT)
+----+----+------+-------------------------------------
```

For rascsi I don't see a straightforward solution because the parser expects each device definition to end with a filename (not starting with a -, i.e. not a command line option). The rasctl parser works differently: filenames are provided with the -f option.
Since the rascsi parser always expects a filename at least a dummy name also has to be provided for scdp and scbr. The actual name can be any string, e.g.

``
rascsi -ID 4 -t scdp foobar
``

rdmark, 2021-08-20:
@uweseimet thanks for the fix for rasctl, and well noted on the rascsi parser. It's a bit of a bummer, since having an 'empty' CD drive be attached as part of the default suite of devices would be very useful indeed.
Your fix for rasctl allows for some hackish scripting on Pi bootup for instance, though, so this is definitely an improvement!

uweseimet, 2021-08-20:
@rdmark I'm also not very happy with the parser, this is why I created https://github.com/akuker/RASCSI/issues/136 some time ago. But I have not found a better solution yet because supporting multiple sets of options requires a marker (the filename in this case) to signal the end of an option set.

By the way, for an empty CD try this:

```
rascsi -ID 3 -t sccd ""
```

rdmark, 2021-08-20:
Yes, this definitely resolves the usecases that I raised this issue for!
For instance, it is possible to do something like this in rascsi.service for a solid default lineup of devices for a Mac:

> ExecStart=/usr/local/bin/rascsi -ID3 -t sccd "" -ID4 -t scdp "" -ID5 -t schd /home/pi/images/600MB.hda 

Not the most elegant looking line, but it works great!

rdmark, 2021-08-20:
Fixed in e3ffc3e

closed 2021-08-20 by rdmark