PiSCSI/piscsi #199: Unable to insert media into an empty CD drive through the webui

rdmark, 2021-08-20:
# Info

- Which version of Pi are you using:  Zero W
- Which github revision of software:  5f4fa7a
- Which board version: 2.4a
- Which computer is the RaSCSI connected to: Mac SE

# Describe the issue

When you have an 'empty' CD ROM drive attached, the webui still only gives you the option to 'Eject' in the Action column, while the SCSI ID that it occupies is not offered as an option to insert a CD image into.
You can still use rasctl to achieve this, so it is simply a webapp issue:

`rasctl -i 2 -c attach -t sccd`
`rasctl -i 2 -c insert -f /home/pi/images/71.iso`

[image]

uweseimet, 2021-08-21:
Ths worked before, indeed. This issue was caused by rasctl displaying "NO MEDIA (WRITE PROTECTED)" for ejected media also instead of just "NO MEDIA". I fixed this in the uweseimet_develop branch.

In general, the fact that the UI depends on rasctl limits what you can do with it. This is why I created https://github.com/akuker/RASCSI/issues/144, but this is a major UI change.

rdmark, 2021-08-21:
@uweseimet Testing in the newly merged develop code (yay!) 0bd12e9 and can see that functionality is improved now.  An empty CD drive has the Detach action associated with it, which functions correctly to detach it altogether.  In addition, the ID of the empty CD drive is available to choose as the target for another image under Image File Management.

However, the remaining problem is that that the Attach button under Image File Management triggers still triggers an attach command, which fails with a duplicate ID error.  So either the webui needs to be aware of this case and use the insert command, or perhaps the attach command can fall back to attempt to inject when failing with a duplicate ID error?

> Aug 22 00:40:47 raspberrypi RASCSI[412]: [2021-08-22 00:40:47.434] [info] +----+----+------+-------------------------------------
> Aug 22 00:40:47 raspberrypi RASCSI[412]: [2021-08-22 00:40:47.437] [info] | ID | UN | TYPE | DEVICE STATUS
> Aug 22 00:40:47 raspberrypi RASCSI[412]: [2021-08-22 00:40:47.437] [info] +----+----+------+-------------------------------------
> Aug 22 00:40:47 raspberrypi RASCSI[412]: [2021-08-22 00:40:47.437] [info] |  1 |  0 | SCCD | /home/pi/images/SWH_LIFE.iso (WRITEPROTECT)
> Aug 22 00:40:47 raspberrypi RASCSI[412]: [2021-08-22 00:40:47.437] [info] |  2 |  0 | SCCD | NO MEDIA
> Aug 22 00:40:47 raspberrypi RASCSI[412]: [2021-08-22 00:40:47.437] [info] +----+----+------+-------------------------------------
> Aug 22 00:40:56 raspberrypi RASCSI[412]: [2021-08-22 00:40:56.406] [info] Validating: cmd=ATTACH, id=2, unit=0, type=SCCD, filename='/home/pi/images/71.iso', device name='', params=''
> Aug 22 00:40:56 raspberrypi RASCSI[412]: [2021-08-22 00:40:56.408] [warning] Duplicate ID 2

uweseimet, 2021-08-22:
Regarding the duplicate ID I'm not so sure whether the previous behavior was correct. When attaching a device the old (master branch) rascsi version ignores an existing device and just overwrites it with the new device. This means that you can, for instance, replace a CD-ROM by a hard disk drive etc. This can lead to issues, because the computer is not prepared for that. It's like disconnecting a physical SCSI device while the computer is up and replacing it with a different device. In practice this leads to undefined and potentially dangerous results. Loss of data, or a computer crash because the driver used for the initial device will usually not work when there is suddenly a different device for the same SCSI ID.
Of course, even with rascsi checking for duplicate IDs you can trigger a scenario like that by simply restarting rascsi with a different setup without rebooting your computer.

So IMO there should be no fallback to INSERT if you ATTACH a device to an ID which is already used, and a DETACH of the previous device should be mandatory.

rdmark, 2021-08-22:
@uweseimet Your reasoning makes complete sense, and I can see the danger in implementing a fallback in the ATTACH command in rascsi/rasctl.  Thanks for going back and comparing with the previous implementation.

rdmark, 2021-08-22:
Actually, I found that there is just a small bug in webui code that's preventing this from work as expected. Will prep a PR momentarily.

rdmark, 2021-08-22:
I think this should do it. Please review! https://github.com/akuker/RASCSI/pull/206
I've tested that it's not possible through the webui to attach an HD or DP device on top of an empty CD device.

closed 2021-08-30 by rdmark