PiSCSI/piscsi #207: Make network interface to use for bridge configurable

uweseimet, 2021-08-23:
It should be possible to provide a comma-separated list of network interfaces to use for setting up the bridge. rascsi should use the first available interface. If "eth0,wlan0,eth1" is requested rascsi should try to use eth0 if available, otherwise wlan0, and if also not available eth1.

uweseimet, 2021-08-24:
@rdmark I created the feature_bridge_interfaces branch for this change and tested it for a Pi with out wlan0 up. Can't currently test it with eth0.

Would be great if you could review and test this feature branch. With debug level trace you can see which devices are being checked. Instead of an empty filename ("") there is actually a parameter for the scdp device now, namely the interface list e.g.

```
rascsi -ID 6 -t scdp eth0,wlan0,eth1
```

The default when no parameter ("") is passed is "eth0,wlan0".

rdmark, 2021-08-24:
@uweseimet I just put in an order for a Pi3 to have an eth0 environment to test in. Will do some thorough testing in a week or so!

uweseimet, 2021-08-24:
@rdmark OK! I will most likely temporarily connect my Pi4 to my LAN later, so that I can then also test eth0.

uweseimet, 2021-08-24:
@rdmark I just connected my Pi to the LAN and the bridge based on eth0 was successfully created and the Mac could access the internet. I have merged the respective feature branch to uweseimet_develop, so that uweseimet_develop contains all my changes.
What I noticed is that when the bridge for eth0 has been set up I cannot remotely (from my PC) connect to my Pi anymore, but I guess this is normal?

closed 2021-09-15 by uweseimet