PiSCSI/piscsi #208: Retranslate Japanese docs

rdmark, 2021-08-24:
The English translations of the documentation under doc/ are in parts incorrect and hard to read.  One of the biggest issues is that the translation is treating line endings as sentence separators, which is incorrect.  There are dozens of Japanese sentences that have been broken up and translated into two unintelligible phrases in English.  I think the best course of action is to start over from scratch and translate the original texts one by one.

Affecting these files:
- doc/converter.txt
- doc/rascsi.txt
- doc/x68k.txt

rdmark, 2021-08-24:
Just read in the [wiki ](https://github.com/akuker/RASCSI/wiki) that they were machine translated by way of Google Translate, which is what I suspected. :)

uweseimet, 2021-08-24:
@rdmark I have to admit, I have never looked at these docs ;-).

Maybe it's worth to translate the original texts with Google Translate again? My guess is that the docs were translated a year ago, and Google Translate may have improved in the meantime.

rdmark, 2021-08-25:
Putting this through Google Translate again after removing all the line endings that are used for formatting would be a huge improvement probably.
I'm pretty committed to translating this by hand now, though. :)

rdmark, 2021-09-15:
As per the discussion in #20, I'm making the goal of this ticket to evaluate the information in the Japanese docs and integrate that information into the wiki.  Afterwards, remove the doc text files from the repo.

rdmark, 2021-09-21:
Wiki documentation created for information that wasn't already in manpages or elsewhere on the wiki.
New page: https://github.com/akuker/RASCSI/wiki/X68000