PiSCSI/piscsi #220: Make reserved host ID dynamically configurable

rdmark, 2021-08-30:
As of develop 62f6c8d, there is hard coding in rascsi-web ractl_cmds.py and templates/index.html to make SCSI ID 7 reserved for the host machine, making it geared towards use on the Mac.
However, other platforms such as SGI (id 0) or Akai samplers (id 6) have different reserved ids.  Platforms like the Atari ST has no reserved id.

While non-Mac users can still use rascsi and rasctl to manage their devices freely, it would be a useful feature for them to have a dynamically configured reserved id to make full use of rascsi-web.

uweseimet, 2021-08-30:
Some notes on the Atari:

- Most ACSI host adapters (used to connect SCSI peripherals to the ACSI bus of the ST or TT) have a SCSI ID, in order to provide SCSI initiator identification, which is required by some SCSI peripherals. These adapters use ID 7 as hard-coded initiator ID.
- For the SCSI bus Atari TT and Falcon store the SCSI initiator ID in their NVRAM. Most drivers for SCSI peripherals (but not Atari's AHDI driver) use this ID for initiator identification. The HDDRIVER driver also supports bus arbitration based on this ID. The ID is configurable, with ID 7 being the default.

rdmark, 2021-08-30:
A first implementation here https://github.com/akuker/RASCSI/pull/223

rdmark, 2021-08-31:
Updated the implementation to support reserving multiple ids.

closed 2021-08-31 by rdmark